OpenDroneMap/ODM #757: Image coordinates for GCP

gpsman, 2018-02-01:
Hi
New in ODM, so sorry if the question sound absurd
- Where is the 0,0 of the image: top left or lower left?
- are decimal pixel coordinates allowed or do we need to use only integer for image coordinate
- is there some tool to extract the image coordinate
Thanks

yjmenezes, 2018-02-01:
@gpsman 

0,0=top-left, x+right, y+dow and integers.  ( i think real would be better )

i upload to my git a small dataset showing GCP files.
https://github.com/yjmenezes/odm_br

I use QGIS to measure the GCPs over the images.

gpsman, 2018-02-01:
Thanks.
I use QGIS too

smathermather, 2018-02-11:
AFAIK, it will accept real numbers. We have a GCP interface for building these that's reaching completion:
https://github.com/posm/posm-gcpi

It's probably already usable enough.

yjmenezes, 2018-02-14:
@smathermather 

inside OpenDromeMap/modules/odm_georef/Georef.hpp

struct GeorefGCP
    int pixelX_;            /**< The pixels x-position for the GCP in the corresponding image **/
    int pixelY_;            /**< The pixels y-position for the GCP in the corresponding image **/

( i think double would be better )

smathermather, 2018-02-15:
Ah, look at that. Yes, we should fix that. Can you submit a pull request, @yjmenezes?

pierotofy, 2019-06-21:
Closing due to inactivity.

closed 2019-06-21 by pierotofy