OpenDroneMap/ODM #759: ExifTools tags and processing

spifftek70, 2018-02-02:
Can you create a list of Exif GPS tags that are required for ODM to properly run and/or enhance the process?   I've search and search, but I can't seem to find a list of hard tags the ODM looks for in an image.

dakotabenjamin, 2018-02-02:
This is a great idea. Here's where in the code ODM extracts exif info: https://github.com/OpenDroneMap/OpenDroneMap/blob/master/opendm/types.py#L68

I will also add this to the documentation.

spifftek70, 2018-02-02:
Thanks for the quick reply!

spifftek70, 2018-02-02:
Hmm..  Have you thought about incorporating `GPSImgDirection` (heading) into the processing?  Might be able to speed things up.

yjmenezes, 2018-02-04:
@spifftek70 

I wrote a bash script to search OpenDroneMap sources tree.
usage:  
find_source_code.sh  ~/OpenDroneMap  py    exiv2 
find_source_code.sh  ~/OpenDroneMap  cpp   exiv2 

output: 

-------- OpenDroneMap/opendm/types.py
import pyexiv2
        self.parse_pyexiv2_values(self.path_file, force_focal, force_ccd)
    def parse_pyexiv2_values(self, _path_file, _force_focal, _force_ccd):
        metadata = pyexiv2.ImageMetadata(_path_file)
            # try/catch tag value due to weird bug in pyexiv2 
            except (pyexiv2.ExifValueError, ValueError) as e:

-------- OpenDroneMap/SuperBuild/src/opensfm/opensfm/geotag_from_gpx.py
import pyexiv2
from pyexiv2.utils import make_fraction
Requires pyexiv2, see install instructions at http://tilloy.net/dev/pyexiv2/
(or use your favorite installer, e.g. 'brew install pyexiv2').
    metadata = pyexiv2.ImageMetadata(filename)
    metadata = pyexiv2.ImageMetadata(filename)

-------- OpenDroneMap/modules/odm_extract_utm/src/UtmExtractor.cpp
#include <exiv2/exiv2.hpp>

I am facing exif tags problems with my church_dataset.

dakotabenjamin, 2018-03-17:
Sorry for the late reply @spifftek70 That sounds like a great addition, and we always welcome community contributions.

skorasaurus, 2019-08-24:
Dakota, I had spent a couple hours debugging a setup only to figure out missing EXIF tags were part of my problem; I'd be glad to make a PR for this but I'm unsure 

cc https://github.com/OpenDroneMap/ODM/issues/910

pierotofy, 2020-09-11:
@skorasaurus a bit late :smile: but you can just add it as a new section ("Exif Tags" or something similar) if you're still interested in making a PR.

fabioandraderj, 2020-09-22:
Do you guys know how I can see a list of EXIF tags used by ODM? 

I tried the types.py file but couldn't find any reference to the GPS tags for example .

It would be great to know the tags that are used to enhance the results as I insert the EXIF data myself in the pictures. That's because I am using a simulation environment with Airsim and Unreal Engine 4 to capture the images.

smathermather, 2020-09-22:
Ha! Pretty interesting use case. Can you come over to community.opendronemap.org for questions?

pierotofy, 2020-09-23:
(Parsing code is now in https://github.com/OpenDroneMap/ODM/blob/master/opendm/photo.py#L94). Yes, questions should go on the forum :pray:

closed 2020-09-23 by smathermather