OpenDroneMap/ODM #760: GCPs detection and ortho-correction

spifftek70, 2018-02-02:
First, I can't say enough on the application.  It's great.

I'm looking to create highly accurate ortho-stitched GeoTiffs (centimeter) using drone images over large fields.  The drone has GPS, but as we all know, they aren't the most accurate of process on small drones.  

We have standardized surveyed GCPs in these fields.  I know that Photoscan has the ability to recognize certain patterns of GCPs and use the corresponding surveyed coordinates to produce highly accurate GeoTiffs.  

Does ODM have this capability or are there plans for such a function?

pierotofy, 2018-02-09:
ODM has supports for GCPs, see https://github.com/OpenDroneMap/OpenDroneMap/wiki/Running-OpenDroneMap#running-odm-with-ground-control but doesn't have automatic pattern recognition built-in yet. There's a project that has started tackling this issue: https://github.com/wolkstein/OpenDroneMap-GCP_LIST.TXT-generator

smathermather, 2018-02-11:
This is something that interests me a lot -- it just needs funding or a champion.

danbjoseph, 2018-02-11:
I had experimented with [wolkstein/OpenDroneMap-GCP_LIST.TXT-generator](https://github.com/wolkstein/OpenDroneMap-GCP_LIST.TXT-generator) but couldn't get it to do a good enough job picking out all the GCPs and it had an even harder time accurately distinguishing the patterns apart.

[image]
[image]

dakotabenjamin, 2018-03-28:
This issue was moved to http://community.opendronemap.org/t/gcps-detection-and-ortho-correction/359

closed 2018-03-28 by dakotabenjamin