OpenDroneMap/ODM #761: Process by list of imagery

spifftek70, 2018-02-05:
I recommend adding a Run Time Parameter of `--image_list` that allows an array of images/paths to be passed to the application for processing.

Rather than only using the Run Time Parameter of `--images` to pass the path to the directory of images, another method would be to pass a list of images for processing.  Sometimes you don't want or need all images in that directory to be processed.  Sometimes you have smaller areas within that collection that you want stitched separately.

yjmenezes, 2018-02-05:
@spifftek70 

I agree, maybe a text file with image { names  [ projection_center  [Omega Phi Kappa | Roll Pith Yaw] ] }
#hash for comments
 like this:

EPSG:32723
#PhotoID E N H 
01_001.jpg 385147.668	9057638.894 2785.308	
01_002.jpg 386280.807 9057648.535 2784.890 

This could be useful for images without exif tags.

pierotofy, 2021-04-05:
Closing, not enough interest and none contributed a PR. Feel free to open a PR if this feature is wanted/needed. :pray:

closed 2021-04-05 by pierotofy