OpenDroneMap/ODM #762: DEM renaming error

thomas-bamford, 2018-02-15:
Hello,

The latest docker image has a renaming error in the DEM generation. This was not occurring in previous versions. Without running DTM and DSM generation no errors occur.

Command used to run ODM in docker, where $1 is the absolute path to the project directory:

docker run -it --rm \
        -v $1/images:/code/images \
        -v $1/odm_georeferencing:/code/odm_georeferencing \
        -v $1/odm_orthophoto:/code/odm_orthophoto \
        -v $1/odm_dem:/code/odm_dem \
        opendronemap/opendronemap --resize-to -1 --dtm --dsm --orthophoto-resolution 80.0

Error during DEM generation:

[DEBUG]   running l2d_dems dtm /code/odm_dem --slope 10 --cellsize 2  -o -s /code/odm_georeferencing/odm_georeferenced_model.bounds.shp --outdir /code/odm_dem --maxsd 2.5 --maxangle 20 --resolution 0.1 --radius 0.1 0.3 0.9 2.7  --gapfill 
Creating odm_dem/odm_georeferenced_model.bounds-0_dtm_r0.1 [idw] from 1 files
Completed odm_dem/odm_georeferenced_model.bounds-0_dtm_r0.1 [idw] in 0:00:15.319531
Creating odm_dem/odm_georeferenced_model.bounds-0_dtm_r0.3 [idw] from 1 files
Completed odm_dem/odm_georeferenced_model.bounds-0_dtm_r0.3 [idw] in 0:00:16.200443
Creating odm_dem/odm_georeferenced_model.bounds-0_dtm_r0.9 [idw] from 1 files
Completed odm_dem/odm_georeferenced_model.bounds-0_dtm_r0.9 [idw] in 0:00:23.187186
Creating odm_dem/odm_georeferenced_model.bounds-0_dtm_r2.7 [idw] from 1 files
Completed odm_dem/odm_georeferenced_model.bounds-0_dtm_r2.7 [idw] in 0:01:25.305160
Completed gap-filling to create odm_dem/odm_georeferenced_model.bounds-0_dtm.idw.tif in 0:00:01.117341
l2d_dems dtm completed (odm_dem) in 0:02:21.159308
Traceback (most recent call last):
  File "/code/run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/odm_dem.py", line 142, in process
    os.rename(os.path.join(odm_dem_root, dtm_pattern), dtm_output_filename)
OSError: [Errno 2] No such file or directory

Radiognome1971, 2018-02-17:
Just got exactly the same error...

pierotofy, 2018-02-17:
This should be fixed in #743. The PR is still waiting review.

pierotofy, 2018-03-02:
This should be fixed now. if not please reopen this issue. Thanks!

closed 2018-03-02 by pierotofy