OpenDroneMap/ODM #765: Mapir Survey 2

LucasMM86, 2018-02-20:
I am using the camera mapir survey 2 and am not getting very good results. The point cloud doesnt look like a nice surface, the mesh has bubbles and consecuently the orthophoto is not good. I think this is because of the distortion.
The manufacturer shows some correction parameters (https://www.mapir.camera/blogs/guide/90677894-adding-camera-profiles-to-pix4d-to-improve-processing-results):

<cameraModelDB><version>17</version><rigs/><cameras><camera name="MAPIR_Survey2_RGB_4.0_4608x3456"><imageWidth>4608</imageWidth><imageHeight>3456</imageHeight><pixelSize>1.34</pixelSize><principalPointXmm>3.08736</principalPointXmm><principalPointYmm>2.31552</principalPointYmm><lensType>perspective</lensType><focalLengthmm>4.06619</focalLengthmm><distortion>5</distortion><radialK1>-0.124322</radialK1><radialK2>0.126914</radialK2><radialK3>0.000642454</radialK3><tangentialT1>0.000415517</tangentialT1><tangentialT2>0.000314419</tangentialT2><cameraModelSource>user</cameraModelSource><bandConfig><band name="Red" centralWaveLength="660" width="0" weight="0.2126"/><band name="Green" centralWaveLength="550" width="0" weight="0.7152"/><band name="Blue" centralWaveLength="470" width="0" weight="0.0722"/></bandConfig></camera></cameras><mappings><mapping exif="Survey2_RGB_4.0_4608x3456" camera="MAPIR_Survey2_RGB_4.0_4608x3456"/></mappings></cameraModelDB>

Can i configure ODM to use this parameters?
I remember there is a file somewhere with correction parameters for a lot of cameras... I wonder if it is possible to add the parameters for mapir survey.
Maybe this is a question for openSfM?

LucasMM86, 2018-02-21:
Hello,
i tried with the camera calibration script of Dakota (https://github.com/dakotabenjamin/cameracalibration) and now it is much better.
But know i still have some kind of splitted surfaces on the point cloud. It is like a group of points forming a piece of surface in the wrong place, separated from the real surface. This is an example:
[image]

Does someone know this effect and how to correct it?

PeterSprague, 2018-03-19:
On Survey2, are you including --force-focal 3.97 --force-ccd 4.63104?  Flip values if have cameras mounted across flight path.  On my Solos, cameras are mounted with width in line with flight path.

Have you tried tweaking your settings.yaml?

Not sure on your environment, but here is a gist of my settings for forest surveys: https://gist.github.com/PeterSprague/ff8c8f354c24d36658208decb6d8f8ad

Likely still needs some tuning, but should get you started. 

On Survey2, are you including --force-focal 3.97 --force-ccd 4.63104?  Flip values if have cameras mounted across flight path.  On my Solos, cameras are mounted with width in line with flight path.

dakotabenjamin, 2018-03-28:
This issue was moved to http://community.opendronemap.org/t/mapir-survey-2/361

closed 2018-03-28 by dakotabenjamin