OpenDroneMap/ODM #767: Error in open cloud points

Leonardojoao01, 2018-02-22:
I am having problem to open the cloud points file, gives the following error:

While opening: .../odm_meshing/odm_mesh.ply'
Error encountered while loading file: .../odm_meshing/odm_mesh.ply"
Error details: Unespected eof

Anyone know the reason for the 3d file when opening becomes all black? Looks like the texture is missing.

The orthosauric is generating correctly.

yjmenezes, 2018-02-22:
@Leonardojoao01 

What SW are you using to open *ply ?
Meshlab, CloudCompare ?

Leonardojoao01, 2018-02-22:
Meshlab

yjmenezes, 2018-02-22:
@Leonardojoao01 

Meshlab works for  me with outputs from ODM 0.3.1 running under Ubuntu1604.
Try reprocess yours dataset.

Leonardojoao01, 2018-02-24:
I installed the CloudCompare solved the problem.

Can you tell me what parameter I need to change to change the density of the point cloud? and resolution map 3d ?

yjmenezes, 2018-02-25:
@Leonardojoao01 

Sorry, i am not testing point cloud at moment. I have to test it too.
try follow this issue: #769
runtime options: https://github.com/OpenDroneMap/OpenDroneMap/wiki/Run-Time-Parameters

closed 2018-03-16 by smathermather