OpenDroneMap/ODM #768: [ERROR]   Not enough supported images in /code/images

hbzhang, 2018-02-22:
Environment, ubuntu 16.04 64bit. 

INFO]    Initializing OpenDroneMap app - Thu Feb 22 21:47:10  2018
[INFO]    Running ODM Load Dataset Cell
[DEBUG]   Loading dataset from: /code/images
[DEBUG]   /code/images
[ERROR]   Not enough supported images in /code/images
[INFO]    OpenDroneMap app finished - Thu Feb 22 21:47:10  2018

Thanks!!

hbzhang, 2018-02-22:
I have found the error,  it seems under images folder, we need to put some images to make it work.

closed 2018-02-24 by pierotofy

r4space, 2018-03-12:
I also have this error and there are plenty of images in the folder...?

Also ubuntu 16.04 but can't get it to install from source so tried docker but no luck either :(

Ran exactly:
docker run -it --rm -v $(pwd)/images:/code/images -v $(pwd)/odm_orthophoto:/code/odm_orthophoto -v $(pwd)/odm_texturing:/code/odm_texturing opendronemap/opendronemap

$(pwd)/images/ is a whole flight's worth of geotiffs

:?

bailingxiao, 2018-04-24:
I also encountered this error, if anyone can solve it, please help me, thanks！

smathermather, 2018-04-24:
@r4space and @bailingxiao -- OpenDroneMap doesn't run on geotiffs, that's why it isn't running on your images. What are the images and what are you trying to accomplish?

bailingxiao, 2018-04-24:
@smathermather 
Thank you very much for your help!My images are download from here https://github.com/OpenDroneMap/odm_data. It's JPG.
I run using docker with the command "docker run -it --rm -v "$(pwd)/images:/code/images" -v "$(pwd)/odm_orthophoto:/opendronemap/opendronemap",there will be an error "Not enough supported images in /code/images". But there are lots of images,and what is the "code/images"?i can't find it.

smathermather, 2018-04-24:
Hey @bailingxiao -- can we move this to http://community.opendronemap.org? That's a better location for support questions. Thanks!

grumpydev-net, 2018-09-15:
How was this closed without providing a solution? And there is no relevant conversation in the suggested community page.  I'm having the same issues, hitting the right folder (otherwise it says it is creating the folder for you), filled with 150+ JPEGs. R permissions to everybody, running in Docker for Mac.

smathermather, 2018-09-15:
Post the question at community.opendronemap.org then and we’ll try to answer. GitHub isn’t a good place for support (better for bugs and other activities closer to the code), but our community page is!

xx1425942154, 2023-09-19:
There are images in my images directory, but he still threw this error. Please help me to answer it. Thank you

smathermather, 2023-09-19:
> There are images in my images directory, but he still threw this error. Please help me to answer it. Thank you

Post the question at community.opendronemap.org then and we’ll try to answer. GitHub isn’t a good place for support (better for bugs and other activities closer to the code), but our community page is!