OpenDroneMap/ODM #769: Expose some opensfm depthmaps config parameters

pierotofy, 2018-02-23:
I can take care of this sometimes in the next few weeks (or if somebody else beats me to it).

I propose to expose some of opensfm's depthmaps parameters since I think the defaults sometimes can lead to less than optimal results. In particular, to increase the quality of our meshes and orthophotos, we could really benefit from getting more points in the dense reconstruction. Here are some of my tests with certain parameters:

`depthmap_min_consistent_views`:

[image]

Default: 3 ( ~262k points )
Changed: 2 ( ~515k points )
Notes: more noise

`depthmap_resolution`:

[image]

Default: 640 ( ~262k points )
Changed: 1280 ( ~1 million points )
Notes: slower by a factor of ~4

`depthmap_patchmatch_iterations`: virtually no change in output, just slower (increased `3` to `9`).

`depthmap_min_patch_sd`: 

Default: 5 ( ~262k points )
Changed: 2.5 ( ~310k points )
Notes: roads appear! same runtime

[image]

`depthmap_same_depth_threshold`: virtually no change in output (increased `0.005` to `0.015`)

`depthmap_num_matching_views`: virtually no change in output (increased `2` to `4`)

`depthmap_method`:

[image]

Default: PATCH_MAP ( ~262k points )
Changed: BRUTE_FORCE( ~330k points )
Notes: Less noise, roads appear, runtime 3x slower

pierotofy, 2018-02-23:
Overall I think we could expose:
 - depthmap_min_consistent_views
 - depthmap_method
 - depthmap_min_patch_sd

As they seem to impact the output the most.

smathermather, 2018-03-16:
Love this! I've been playing with these parameters for specific projects, and would love to see them exposed. I would expose ```depthmap_resolution``` which defaults to a pretty conservative value.

pierotofy, 2018-05-15:
I'll try to get this done during the code sprint this week.

closed 2018-08-15 by pierotofy