OpenDroneMap/ODM #770: Can not build the docker file

hbzhang, 2018-02-24:
On Ubuntu 16.04

The following can not pass

git clone https://github.com/OpenDroneMap/OpenDroneMap.git
cd OpenDroneMap
docker build -t my_odm_image .

It complains the following error:

Your branch is up-to-date with 'origin/master'.
[ 45%] No patch step for 'ext_mve'
[ 47%] No update step for 'ext_mve'
[ 49%] No configure step for 'ext_mve'
[ 50%] Performing build step for 'ext_mve'
[ 52%] No install step for 'ext_mve'
[ 54%] Completed 'ext_mve'
[ 54%] Built target ext_mve
Scanning dependencies of target tex
[ 55%] Building CXX object libs/tex/CMakeFiles/tex.dir/build_adjacency_graph.cpp.o
[ 57%] Building CXX object libs/tex/CMakeFiles/tex.dir/build_obj_model.cpp.o
[ 59%] Building CXX object libs/tex/CMakeFiles/tex.dir/calculate_data_costs.cpp.o
[ 61%] Building CXX object libs/tex/CMakeFiles/tex.dir/generate_debug_embeddings.cpp.o
[ 62%] Building CXX object libs/tex/CMakeFiles/tex.dir/generate_texture_atlases.cpp.o
[ 64%] Building CXX object libs/tex/CMakeFiles/tex.dir/generate_texture_patches.cpp.o
[ 66%] Building CXX object libs/tex/CMakeFiles/tex.dir/generate_texture_views.cpp.o
[ 67%] Building CXX object libs/tex/CMakeFiles/tex.dir/global_seam_leveling.cpp.o
[ 69%] Building CXX object libs/tex/CMakeFiles/tex.dir/histogram.cpp.o
[ 71%] Building CXX object libs/tex/CMakeFiles/tex.dir/local_seam_leveling.cpp.o
[ 72%] Building CXX object libs/tex/CMakeFiles/tex.dir/material_lib.cpp.o
[ 74%] Building CXX object libs/tex/CMakeFiles/tex.dir/obj_model.cpp.o
[ 76%] Building CXX object libs/tex/CMakeFiles/tex.dir/poisson_blending.cpp.o
[ 77%] Building CXX object libs/tex/CMakeFiles/tex.dir/prepare_mesh.cpp.o
[ 79%] Building CXX object libs/tex/CMakeFiles/tex.dir/rectangular_bin.cpp.o
[ 81%] Building CXX object libs/tex/CMakeFiles/tex.dir/seam_leveling.cpp.o
[ 83%] Building CXX object libs/tex/CMakeFiles/tex.dir/texture_atlas.cpp.o
[ 84%] Building CXX object libs/tex/CMakeFiles/tex.dir/texture_patch.cpp.o
[ 86%] Building CXX object libs/tex/CMakeFiles/tex.dir/texture_view.cpp.o
[ 88%] Building CXX object libs/tex/CMakeFiles/tex.dir/timer.cpp.o
[ 89%] Building CXX object libs/tex/CMakeFiles/tex.dir/tri.cpp.o
[ 91%] Building CXX object libs/tex/CMakeFiles/tex.dir/uni_graph.cpp.o
[ 93%] Building CXX object libs/tex/CMakeFiles/tex.dir/view_selection.cpp.o
/code/SuperBuild/src/mvstexturing/libs/tex/view_selection.cpp: In function ‘void tex::view_selection(const DataCosts&, UniGraph*, const tex::Settings&)’:
/code/SuperBuild/src/mvstexturing/libs/tex/view_selection.cpp:62:67: error: invalid conversion from ‘mapmap::Graph<float>*’ to ‘mapmap::luint_t {aka long unsigned int}’ [-fpermissive]
     mapmap::UnaryTable<cost_t, simd_w> unaries(&mgraph, &label_set);
                                                                   ^
In file included from /code/SuperBuild/src/mvstexturing/elibs/mapmap/mapmap/header/cost_instances/unary_table.h:50:0,
                 from /code/SuperBuild/src/mvstexturing/elibs/mapmap/mapmap/source/optimizer_instances/dp_node_solver_factory.impl.h:12,
                 from /code/SuperBuild/src/mvstexturing/elibs/mapmap/mapmap/header/optimizer_instances/dp_node_solver_factory.h:34,
                 from /code/SuperBuild/src/mvstexturing/elibs/mapmap/mapmap/source/optimizer_instances/dynamic_programming.impl.h:18,
                 from /code/SuperBuild/src/mvstexturing/elibs/mapmap/mapmap/header/optimizer_instances/dynamic_programming.h:112,
                 from /code/SuperBuild/src/mvstexturing/elibs/mapmap/mapmap/header/mapmap.h:23,
                 from /code/SuperBuild/src/mvstexturing/elibs/mapmap/mapmap/full.h:18,
                 from /code/SuperBuild/src/mvstexturing/libs/tex/view_selection.cpp:14:
/code/SuperBuild/src/mvstexturing/elibs/mapmap/mapmap/source/cost_instances/unary_table.impl.h:26:1: note:   initializing argument 1 of ‘mapmap::UnaryTable<COSTTYPE, SIMDWIDTH>::UnaryTable(mapmap::luint_t, const mapmap::LabelSet<COSTTYPE, SIMDWIDTH>*) [with COSTTYPE = float; unsigned int SIMDWIDTH = 8u; mapmap::luint_t = long unsigned int]’
 UnaryTable<COSTTYPE, SIMDWIDTH>::
 ^
/code/SuperBuild/src/mvstexturing/libs/tex/view_selection.cpp:79:17: error: ‘class mapmap::UnaryTable<float, 8u>’ has no member named ‘set_costs_for_node’
         unaries.set_costs_for_node(i, costs);
                 ^
/code/SuperBuild/src/mvstexturing/libs/tex/view_selection.cpp:90:55: error: wrong number of template arguments (4, should be 2)
     mapmap::mapMAP<cost_t, simd_w, unary_t, pairwise_t> solver;
                                                       ^
In file included from /code/SuperBuild/src/mvstexturing/elibs/mapmap/mapmap/full.h:18:0,
                 from /code/SuperBuild/src/mvstexturing/libs/tex/view_selection.cpp:14:
/code/SuperBuild/src/mvstexturing/elibs/mapmap/mapmap/header/mapmap.h:59:7: note: provided for ‘template<class COSTTYPE, unsigned int SIMDWIDTH> class mapmap::mapMAP’
 class mapMAP
       ^
/code/SuperBuild/src/mvstexturing/libs/tex/view_selection.cpp:91:12: error: request for member ‘set_graph’ in ‘solver’, which is of non-class type ‘int’
     solver.set_graph(&mgraph);
            ^
/code/SuperBuild/src/mvstexturing/libs/tex/view_selection.cpp:92:12: error: request for member ‘set_label_set’ in ‘solver’, which is of non-class type ‘int’
     solver.set_label_set(&label_set);
            ^
/code/SuperBuild/src/mvstexturing/libs/tex/view_selection.cpp:93:12: error: request for member ‘set_unaries’ in ‘solver’, which is of non-class type ‘int’
     solver.set_unaries(&unaries);
            ^
/code/SuperBuild/src/mvstexturing/libs/tex/view_selection.cpp:94:12: error: request for member ‘set_pairwise’ in ‘solver’, which is of non-class type ‘int’
     solver.set_pairwise(&pairwise);
            ^
/code/SuperBuild/src/mvstexturing/libs/tex/view_selection.cpp:95:12: error: request for member ‘set_logging_callback’ in ‘solver’, which is of non-class type ‘int’
     solver.set_logging_callback(display);
            ^
/code/SuperBuild/src/mvstexturing/libs/tex/view_selection.cpp:96:12: error: request for member ‘set_termination_criterion’ in ‘solver’, which is of non-class type ‘int’
     solver.set_termination_criterion(&terminate);
            ^
/code/SuperBuild/src/mvstexturing/libs/tex/view_selection.cpp:99:12: error: request for member ‘optimize’ in ‘solver’, which is of non-class type ‘int’
     solver.optimize(solution);
            ^
libs/tex/CMakeFiles/tex.dir/build.make:590: recipe for target 'libs/tex/CMakeFiles/tex.dir/view_selection.cpp.o' failed
make[5]: *** [libs/tex/CMakeFiles/tex.dir/view_selection.cpp.o] Error 1
CMakeFiles/Makefile2:278: recipe for target 'libs/tex/CMakeFiles/tex.dir/all' failed
make[4]: *** [libs/tex/CMakeFiles/tex.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make[3]: *** [all] Error 2
CMakeFiles/mvstexturing.dir/build.make:113: recipe for target 'mvstexturing/stamp/mvstexturing-build' failed
make[2]: *** [mvstexturing/stamp/mvstexturing-build] Error 2
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/mvstexturing.dir/all' failed
make[1]: *** [CMakeFiles/mvstexturing.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
The command '/bin/sh -c cd SuperBuild && mkdir build && cd build && cmake  .. && make -j$(nproc)     && cd ../.. && mkdir build && cd build && cmake .. && make -j$(nproc)' returned a non-zero code: 2

pierotofy, 2018-02-24:
I've also had this problem two days ago trying to build it on a MacOS machine. Haven't had time to find out why.

pierotofy, 2018-02-25:
Once https://github.com/OpenDroneMap/mvs-texturing/pull/2 is merged, we can open a PR to include the new changes to mvs-texturing to fix the build.

In the meanwhile for those that can't wait, just replace `External-MvsTexturing.cmake:11` with:

```
URL               https://github.com/pierotofy/mvs-texturing/archive/mvs-update.zip
#URL_MD5           cbcccceba4693c6c882eb4aa618a2227
```

Don't forget the # in front of URL_MD5.

@dakotabenjamin

hbzhang, 2018-02-26:
> Once OpenDroneMap/mvs-texturing#2

thanks, will give it a try!

dakotabenjamin, 2018-02-26:
It's been merged.

PR #772 (2018-02-27): Updated mvs-texturing to latest version
Fixes #770 by pulling latest mvs-texturing code.
closed 2018-02-26 by dakotabenjamin