OpenDroneMap/ODM #773: Illegal instruction (core dumped)

krzysiu4, 2018-02-28:
I am trying to create simple orthophotomap, consisting of few example photos, but always get error:

```
[DEBUG]   Writing ODM Mesh file in: /code/odm_meshing/odm_mesh.ply
[DEBUG]   running /code/build/bin/odm_meshing -inputFile /code/opensfm/depthmaps/merged.ply -outputFile /code/odm_meshing/odm_mesh.ply -logFile /code/odm_meshing/odm_meshing_log.txt -maxVertexCount 100000 -octreeDepth 9  -samplesPerNode 1.0 -solverDivide 9
Illegal instruction (core dumped)
Traceback (most recent call last):
  File "/code/run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/odm_meshing.py", line 79, in process
    '-samplesPerNode {samples} -solverDivide {solver}'.format(**kwargs))
  File "/code/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 132
```

 I am running Ubuntu 16.04 and trying to run ODM through Docker. I tried both the latest and wb build without success. I also tried using WebODM, but it gives the same error. I am struggling with this problem for two days and can't find solution. Anyone know how to fix this?

I used few pictures from this example: [Link](https://github.com/OpenDroneMap/odm_data_toledo/tree/master/images)

Thanks in advance.

smathermather, 2018-03-16:
If there aren't further questions, I'll close this issue. Feel free to reopen if necessary.

closed 2018-03-16 by smathermather

PeterSprague, 2018-05-18:
I'm now getting this exact same error after building with latest OpenSfM master.
Fixed the missing unicode errors in io.py of OpenSfM [see https://github.com/mapillary/OpenSfM/issues/325].  Wonder if this is causing odm_mesh errors?

2018-05-18 21:15:36,694 INFO: Pruning depthmap for image 2017_1016_133647_208.JPG
2018-05-18 21:15:48,718 INFO: Merging depthmaps
[DEBUG]   running PYTHONPATH=/usr/local/lib/python2.7/dist-packages /usr/local/src/OpenSfM/bin/export_bundler /var/www/data/ca4e2822-8c35-4d3e-9c44-62b7c4088b8e/opensfm
[DEBUG]   running PYTHONPATH=/usr/local/lib/python2.7/dist-packages /usr/local/src/OpenSfM/bin/opensfm export_geocoords /var/www/data/ca4e2822-8c35-4d3e-9c44-62b7c4088b8e/opensfm --transformation --proj '+units=m +no_defs=True +datum=WGS84 +proj=utm +zone=10 '
[INFO]    Running ODM OpenSfM Cell - Finished
[INFO]    Running ODM Meshing Cell
[DEBUG]   Writing ODM Mesh file in: /var/www/data/ca4e2822-8c35-4d3e-9c44-62b7c4088b8e/odm_meshing/odm_mesh.ply
[DEBUG]   running /code/build/bin/odm_meshing -inputFile /var/www/data/ca4e2822-8c35-4d3e-9c44-62b7c4088b8e/opensfm/depthmaps/merged.ply -outputFile /var/www/data/ca4e2822-8c35-4d3e-9c44-62b7c4088b8e/odm_meshing/odm_mesh.ply -logFile /var/www/data/ca4e2822-8c35-4d3e-9c44-62b7c4088b8e/odm_meshing/odm_meshing_log.txt -maxVertexCount 30000 -octreeDepth 5 -verbose -samplesPerNode 1.0 -solverDivide 9
The number of samples per octree node was manually set to: 1
Numerical solver divisions was manually set to: 5
Illegal instruction (core dumped)
Traceback (most recent call last):
  File "/code/run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/odm_meshing.py", line 80, in process
    '-samplesPerNode {samples} -solverDivide {solver}'.format(**kwargs))
  File "/code/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 132