OpenDroneMap/ODM #775: Visibility-Consistent Thin Surface Reconstruction (TSR)

sberryman, 2018-03-03:
Have any project contributors looked at the TSR paper and source code? The results from the paper look pretty incredible. I know it is a far stretch from the current software used in the pipeline but I don't see why OpenSfM couldn't create linked point clouds to the sensors which captured them. The depth map linking looks a bit more difficult but that is due to the lack of experience with ODM.

Paper:
https://www.gcc.tu-darmstadt.de/media/gcc/papers/Aroudj-2017-TSR-Prepress.pdf

Source code:
https://github.com/SamirAroudj/TSR

Website:
https://www.gcc.tu-darmstadt.de/home/proj/tsr/tsr.en.jsp

pierotofy, 2018-03-03:
Very cool research, thanks for posting the links.

Would be interesting to see how it renders a terrain scene, keeping in mind that most of the improvements this algorithm focuses on improving are thin objects. Maybe you'll get better looking vegetation.

Input requires points with "visibility information", so won't work straight out of OpenSfM.

sberryman, 2018-03-09:
Sorry, thought I followed back up on your comment. Agree that the title alludes to thin objects but it seemed to do a pretty good job on Stephansdom reconstruction.

[image]
[image]
Image source: https://www.gcc.tu-darmstadt.de/home/proj/tsr/tsr.en.jsp

On a side note, I took a lot of your suggestions from #769. Specifically `depthmap_resolution=1280`, `depthmap_min_patch_sd=2.5` and `depthmap_method=BRUTE_FORCE`. I didn't pay a whole lot of attention to your time estimates and now waiting patiently for it to finish. It has been running since March 2nd and is only on `odm_25dmeshing`. The project contains ~1,443 images so I probably went a bit overboard.

pierotofy, 2018-03-09:
Yeah I wouldn't recommend using the --use-25dmesh option for dense reconstructions, it will perform poorly.

smathermather, 2018-03-16:
@sberryman -- why close this issue? I'm normally all for closing issues 😄, but this is an intriguing proposal and could stand to be explored.

sberryman, 2018-03-16:
@smathermather simply to reduce your list of open issues, I figured it was more of an FYI and I didn't know about the gitter channel yet so this was my way of letting you guys know.

smathermather, 2020-04-14:
Still an interesting feature. @adamsteer -- do you want to look into this? :D

adamsteer, 2020-04-15:
I already am, very slowly ;). I have to focus on things I can invoice for (or sell) :/

smathermather, 2020-04-15:
Haha! Understood. :)

smathermather, 2022-03-12:
Going to close this. TSR is interesting, and good meshing is important, but my experience with this implementation is that it is quite difficult to build.

closed 2022-03-12 by smathermather