OpenDroneMap/ODM #776: Completion time estimates

arkadyark, 2018-03-05:
Is anything like this available, or would something like this be possible? Even a rough estimate, like "2-4 hours remaining" would be helpful.

smathermather, 2018-03-16:
This is one of those features that is impossible to build well, but also desirable. I think @pierotofy was looking at the possibility of having a portion completed bar in WebODM, but I don't know if it's still just a wish list item, or something in progress.

pierotofy, 2018-03-17:
We could create probable estimates based on a particular formula, but wouldn't be very accurate. Lots of factors at play and tweaking the parameters can change the estimates considerably.

pierotofy, 2018-03-17:
Mandatory: https://xkcd.com/612/

sberryman, 2018-03-17:
What about showing progress for each sub process? Even watching file counts in a directory can give a good idea of progress and duration for a lot of OpenSFM tasks.

So maybe starting out with a visual checklist? The fact OpenSFM has sub tasks of extract_metadata, detect_features, match_features, create_tracks, reconstruct, mesh, export_visualsfm, undistort, compute_depthmaps, export_bundler and export_geocoords was a surprise to me at first. (I know the pipeline is different based on settings but that is still a lot of sub tasks)

I don't recall seeing that in WebODM but I've moved along to manually executing all the steps to see what is going on.

smathermather, 2020-04-14:
Closing. We now have https://github.com/OpenDroneMap/odm-benchmarks!

closed 2020-04-14 by smathermather