OpenDroneMap/ODM #777: Reconstruction

paulinK, 2018-03-07:
Hi Everyone
I'm trying to process a map with 685 images. But, there is a problem. At the reconstruction level, opendronemap (opensfm) outputs many reconstruction groups and keeps the first one to build the map.
This leads us to bad result. And I would like to know if there is a way to force opendronemap to use all the reconstruction groups defined for the map.
In addition, I wonder if there is an algorithm that could tell us the number of images opendronemap (opensfm) will keep for the first reconstruction (with reference to parameters).

I want to know also, if flight direction doesn't affect the reconstruction. 

This is the parameter I have used (I have modified the opendronemap settings file directly)

---
# A list of global configuration variables
# Uncomment lines as needed to edit default settings.
# Note this only works for settings with default values. Some commands like --rerun <module>
# or --force-ccd n will have to be set in the command line (if you need to)

# This line is really important to set up properly
project_path: '/' #DO NOT CHANGE THIS OR DOCKER WILL NOT WORK. It should be '/'

# The rest of the settings will default to the values set unless you uncomment and change them
resize_to: 4000
#start_with: 'resize'
#end_with: 'odm_orthophoto'
#rerun_all: False
#zip_results: False
#verbose: False
#time: False
#use_fixed_camera_params: False
#use_hybrid_bundle_adjustment: False
#opensfm_processes: 4 # by default this is set to $(nproc)
#min_num_features: 8000
#matcher_threshold: 2.0
#matcher_ratio: 0.6
#matcher_neighbors: 8
#matcher_distance: 0
#use_pmvs: False # The cmvs/pmvs settings only matter if 'Enabled' is set to True
cmvs_maximages: 2000
#pmvs_level: 1
#pmvs_csize: 2
#pmvs_threshold: 0.7
#pmvs_wsize: 7
#pmvs_min_images: 3
#pmvs_num_cores: 4 # by default this is set to $(nproc)
#mesh_size: 100000000
#mesh_octree_depth: 9
#mesh_samples: 10000.0
#mesh_solver_divide: 9
#texturing_data_term: 'gmi'
#texturing_outlier_removal_type: 'gauss_clamping'
#texturing_skip_visibility_test: False
#texturing_skip_global_seam_leveling: False
#texturing_skip_local_seam_leveling: False
#texturing_skip_hole_filling: False
#texturing_keep_unseen_faces: False
#texturing_tone_mapping: 'none'
#gcp: !!null  # YAML tag for None
use_exif: True # Set to True if you have a GCP file (it auto-detects) and want to use EXIF
#dtm: False      # Use this tag to build a DTM (Digital Terrain Model
#dsm: False      # Use this tag to build a DSM (Digital Surface Model
#dem-gapfill-steps: 4
#dem-resolution: 0.1
#dem-maxangle:20
#dem-maxsd: 2.5
#dem-approximate: False
#dem-decimation: 1
#dem-terrain-type: ComplexForest
orthophoto_resolution: 33.33333333 # Pixels/meter
#orthophoto_target_srs: !!null # Currently does nothing
#orthophoto_no_tiled: False
#orthophoto_compression: DEFLATE # Options are [JPEG, LZW, PACKBITS, DEFLATE, LZMA, NONE] Don't change unless you know what you are doing
#orthophoto_bigtiff: IF_SAFER # Options are [YES, NO, IF_NEEDED, IF_SAFER]
#build_overviews: FALSE
#pc-classify: none

Best regards

smathermather, 2018-03-16:
Hi @paulinK -- what kind of camera are you working with?

paulinK, 2018-03-16:
Hi @smathermather  
I have a DJI camera (Phantom 4).

smathermather, 2018-03-16:
That'll do it. You need to calibrate your images due to barrel distortion before processing them in ODM. OpenSfM, the underlying library, won't address barrel distortion on its own. See https://github.com/OpenDroneMap/CameraCalibration .

closed 2018-03-16 by smathermather

smathermather, 2018-03-16:
Also, I've closed this issue, but feel free to ask more questions or re-open as necessary.