OpenDroneMap/ODM #779: partial othrophoto creation

glw, 2018-03-07:
Im getting a similar output as issue #491 . Where my total AOI is not included in the final orthophoto.
Im using the docker version

`docker run -it --rm -v $(pwd):/code/images -v $(pwd)/odm_orthophoto:/code/odm_orthophoto -v $(pwd)/odm_texturing:/code/odm_texturing opendronemap/opendronemap --fast-orthophoto`

here are my results:
[image]

The points are the exif locations of my imagery. 

My overlap should have been somewhere around 80/80 or 85/85. Camera is Parrot Sequoia and Im only running the RGB images at the moment.

I saw in the previous issue, he had tweaked the `--min_num_features` , is this still the recommended route?

Thanks!

smathermather, 2018-03-16:
@glw -- did you calibrate your parrot images? My hunch is barrel distortion is causing issues in the matching. @dakotabenjamin has a set of scripts for addressing this at https://github.com/OpenDroneMap/CameraCalibration

dakotabenjamin, 2018-03-28:
This issue was moved to http://community.opendronemap.org/t/partial-othrophoto-creation/393

closed 2018-03-28 by dakotabenjamin

kikislater, 2018-04-25:
@glw Do you get decent result with this poor sensor readout on sequoia (Sony IMX206) ?