OpenDroneMap/ODM #782: Georef fail on ODM ('list' object has no attribute 'georef')

briansaab, 2018-03-15:
Please see the following topic: http://community.opendronemap.org/t/georef-fail-on-odm-list-object-has-no-attribute-georef/309

dakotabenjamin, 2018-03-17:
I haven't used PMVS in a while, and was thinking not a lot of users still use that method. I've been thinking of removing it, in fact. Before I look into this bug too deeply I would be interested to know 
- how much better is the PMVS output currently
- if changes to OpenSfM might see enough improvements to warrant removal of the module

It's simpler to maintain one method and I'd like to see everyone transition to OpenSfM, but I want to make sure people who have been using PMVS are happy with that.

briansaab, 2018-03-19:
I find that PMVS does produce smoother, more complete point clouds (without as many gaps) as what comes natively with OpenSFM.  That said, it could just be the standard settings for PMVS are geared towards producing a high quality point cloud (and subsequent DEM/DSM) and I'm just not dialing in the right point cloud parameters for OpenSFM.  Getting OpenSFM to meet the quality bar would be great as I do find that adding PMVS to the process consistently increases the processing time and can lead to memory fault errors (even on a 32 GB RAM device) as it is a hog.

PeterSprague, 2018-03-20:
I am attempting to use PMVS via node-odm.  Reasoning is to try and get higher quality forest survey orthos than I'm getting from OpenSfM.

Getting the exact same error on a set of images that run on OpenSfM.  Tried with -j$(nproc) and -j4 on a 32 core workstation.

briansaab, 2018-03-26:
It's worth noting that I think @dakotabenjamin and @pierotofy are both already tracking issues related to what I'm trying to achieve.  Namely, I'd like to get a denser point cloud and today PMVS is proving to do a better job than Opensfm.

That said, there are already some existing issues opened (#662 & #769) where it looks like the suggestion is to expose more depthmap options related to Opensfm, which would be great and allow us to test Opensfm vs. PMVS more concretely.

Any update on when we might see more depthmap options support, even in ODM command line (vs. also backing into WebODM)?

smathermather, 2020-04-14:
PMVS is removed.

closed 2020-04-14 by smathermather