OpenDroneMap/ODM #784: Coordinates file is not building, no ortho generated

PeterSprague, 2018-03-17:
See: http://community.opendronemap.org/t/mapir-survey3-orthophoto-can-not-be-generated/313/6

Ortho not building, png is mirror image of location features.  Images include GPS location and datum

Building objmodel:
Saving model… done.
Whole texturing procedure took: 31.188s
[INFO] Running ODM Texturing Cell - Finished
[INFO] Running ODM Georeferencing Cell
[DEBUG] None
[WARNING] No coordinates file. Generating coordinates file: /var/www/data/d9e91b26-41b5-4e55-803f-a06c769d0d7e/odm_georeferencing/coords.txt
[WARNING] Could not generate coordinates file. Ignore if there is a GCP file. Error: logFile_ was set to: /var/www/data/d9e91b26-41b5-4e55-803f-a06c769d0d7e/odm_georeferencing/odm_georeferencing_utm_log.txt
Error in OdmExtractUtm:

dakotabenjamin, 2018-03-27:
If there is a single image without GPS in the exif, or if the exif is malformed, that may be the cause. I'll label this a bug for now, perhaps we need to add better support for the mapir cameras. Do you know how those files were georeferenced?

PeterSprague, 2018-03-27:
Were not my images, just tested them.  My understanding is that they were produced with the latest MapIR Survey3 camera that has an attached GPS that reads the position directly to EXIF.

I looked through a random selection of the images' EXIF data, and all seemed fine.  The image highlighted as having an error looked fine to me.

There was also the oddity of ODM producing a mirror image PNG of the landscape.

closed 2020-04-14 by smathermather