OpenDroneMap/ODM #785: dataset is loaded again on every rerun, instead of using cached exif data

fredlllll, 2018-03-18:
basically what the title says.

if i want to rerun a step, the exif data of the images is extracted again. as it is single threaded its super slow and can take a considerable amount of time if you have several thousand images

i would propose caching the exif data and parallelizing the extraction process

sberryman, 2018-03-19:
I noticed this issue over weekend as well. It seems like the general practice is to run all steps/cells in the pipeline but skip cells or that have already been completed.

Following that practice, it appears like a check if `tree.dataset_list` exists should be placed around https://github.com/OpenDroneMap/OpenDroneMap/blob/master/scripts/dataset.py#L69

Maybe a new function to load the existing `tree.dataset_list` and compare it to the `files` array to see if new files are added or existing files deleted. In that case (files added or removed) it "should" error out unless `initial_task_id=0` or `rerun=opensfm` or `rerun-from=opensfm`?

As @fredlllll mentioned, `class ODM_Photo`'s `__init__` function calls an internal function `parse_pyexiv2_values` which is not done in parallel.

The other thought is that each image can have it's own JSON/XML/YAML file of extracted exif data and the `ODM_Photo` class can simply check if the file exists within `__init__`, if it does, load the file and populate all the variables (ccd_width, latitude, longitude, etc) within the class or proceed and load it from exif data. This wouldn't solve the parallel issue but probably easier to implement and retains all class properties. I have no idea at this point what other cells are relying on data from `ODM_Photo`

dakotabenjamin, 2018-03-27:
I'm not sure how to go about this because we're in a half-in/'half-out for having actual classes and objects in the code. Some folx have discussed pickling/serializing but we never came to any concensus. I'd be interested to pick that discussion back up again.

fredlllll, 2018-03-27:
should use a language that also isnt half-in half-out on object orientation ;)

go the object oriented way, it makes things so much cleaner, even in python (although its just half baked objects)

dakotabenjamin, 2018-03-29:
We could parallelize https://github.com/OpenDroneMap/OpenDroneMap/blob/master/scripts/dataset.py#L75 as such:

            photos = Pool().map(
                partial(make_odm_photo, self.params.force_focal, self.params.force_ccd),
                path_files
            )

I like @sberryman suggestions, though. Writing more image information to file should be enough.

pierotofy, 2019-05-23:
This has been implemented some time ago via the `images.json` image database. :hugs:

closed 2019-05-23 by pierotofy