OpenDroneMap/ODM #787: odm_meshing_log.txt not show correct mesh_solver_divide setting used

PeterSprague, 2018-03-19:
Run log snippet:
[DEBUG]   running /code/build/bin/odm_meshing -inputFile /var/www/data/ec344287-8ed5-45c5-972d-1f096919b901/opensfm/depthmaps/merged.ply -outputFile /var/www/data/ec344287-8ed5-45c5-972d-1f096919b901/odm_meshing/odm_mesh.ply -logFile /var/www/data/ec344287-8ed5-45c5-972d-1f096919b901/odm_meshing/odm_meshing_log.txt -maxVertexCount 30000 -octreeDepth 6 -verbose -samplesPerNode 1.0 -solverDivide 9
The number of samples per octree node was manually set to: 1
Numerical solver divisions was manually set to: 6
Successfully loaded 4423321 points with corresponding normals from file.
Octree depth used for reconstruction is: 6
Estimated initial vertex count: 4096

smathermather, 2019-11-14:
Meshing has I think changed a bit since this time. @pierotofy: do we know if this is still a bug?

pierotofy, 2019-11-14:
This no longer applies as there's no `odm_meshing` program anymore.

closed 2019-11-14 by pierotofy