OpenDroneMap/ODM #789: Check if proccess is running

muriloazevedo, 2018-03-22:
I started the process yesterday is running without errors like the image below
[image]

But I don't know if what step it's running and any estimative of the whole proccess. Is there any guide to explane what's ODM doing?

Thanks!

yjmenezes, 2018-03-22:
@muriloazevedo 

you can run the pipeline step by step.
https://github.com/OpenDroneMap/OpenDroneMap/wiki/Run-Time-Parameters

--start-with
--end-with
and measure every step.
time  ./run.sh .....

your screenshot shows it is running first pipeline step:   bin/opensfm

dakotabenjamin, 2018-03-27:
Yes, currently it's a "go get some coffee and wait" kind of deal. We have heard a lot of requests for better progress reporting so it's certainly on the backlog.

dakotabenjamin, 2018-03-28:
This issue was moved to http://community.opendronemap.org/t/check-if-proccess-is-running/358

closed 2018-03-28 by dakotabenjamin