OpenDroneMap/ODM #790: Bash configure.sh install error

parang17, 2018-03-22:
I have tried to install OpenDroneMap, but I got the following an error. Can someone help fixing this issue? Thank you!

[ 17%] Built target odm_extract_utm
[ 21%] Linking CXX executable ../../bin/odm_orthophoto
[ 26%] Linking CXX executable ../../bin/odm_georef
[ 30%] Linking CXX executable ../../bin/odm_texturing
[ 34%] Linking CXX executable ../../bin/odm_meshing
../../../SuperBuild/install/lib/libopencv_highgui.so.2.4.11: undefined reference to `TIFFIsTiled@LIBTIFF_4.0'
../../../SuperBuild/install/lib/libopencv_highgui.so.2.4.11: undefined reference to `TIFFOpen@LIBTIFF_4.0'
../../../SuperBuild/install/lib/libpcl_io.so: undefined reference to `png_set_longjmp_fn'
../../../SuperBuild/install/lib/libopencv_highgui.so.2.4.11: undefined reference to `TIFFReadEncodedStrip@LIBTIFF_4.0'
../../../SuperBuild/install/lib/libopencv_highgui.so.2.4.11: undefined reference to `TIFFSetField@LIBTIFF_4.0'
../../../SuperBuild/install/lib/libopencv_highgui.so.2.4.11: undefined reference to `TIFFWriteScanline@LIBTIFF_4.0'
../../../SuperBuild/install/lib/libopencv_highgui.so.2.4.11: undefined reference to `TIFFGetField@LIBTIFF_4.0'
../../../SuperBuild/install/lib/libopencv_highgui.so.2.4.11: undefined reference to `TIFFScanlineSize@LIBTIFF_4.0'
../../../SuperBuild/install/lib/libopencv_highgui.so.2.4.11: undefined reference to `TIFFSetWarningHandler@LIBTIFF_4.0'
../../../SuperBuild/install/lib/libopencv_highgui.so.2.4.11: undefined reference to `TIFFSetErrorHandler@LIBTIFF_4.0'
../../../SuperBuild/install/lib/libopencv_highgui.so.2.4.11: undefined reference to `TIFFReadEncodedTile@LIBTIFF_4.0'
../../../SuperBuild/install/lib/libopencv_highgui.so.2.4.11: undefined reference to `TIFFReadRGBATile@LIBTIFF_4.0'
../../../SuperBuild/install/lib/libopencv_highgui.so.2.4.11: undefined reference to `TIFFClose@LIBTIFF_4.0'
../../../SuperBuild/install/lib/libopencv_highgui.so.2.4.11: undefined reference to `TIFFRGBAImageOK@LIBTIFF_4.0'
../../../SuperBuild/install/lib/libopencv_highgui.so.2.4.11: undefined reference to `TIFFReadRGBAStrip@LIBTIFF_4.0'
collect2: error: ld returned 1 exit status
modules/odm_orthophoto/CMakeFiles/odm_orthophoto.dir/build.make:371: recipe for target 'bin/odm_orthophoto' failed
make[2]: *** [bin/odm_orthophoto] Error 1
CMakeFiles/Makefile2:280: recipe for target 'modules/odm_orthophoto/CMakeFiles/odm_orthophoto.dir/all' failed
make[1]: *** [modules/odm_orthophoto/CMakeFiles/odm_orthophoto.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
../../../SuperBuild/install/lib/libpcl_io.so: undefined reference to `png_set_longjmp_fn'
collect2: error: ld returned 1 exit status
modules/odm_meshing/CMakeFiles/odm_meshing.dir/build.make:368: recipe for target 'bin/odm_meshing' failed
make[2]: *** [bin/odm_meshing] Error 1
CMakeFiles/Makefile2:225: recipe for target 'modules/odm_meshing/CMakeFiles/odm_meshing.dir/all' failed
make[1]: *** [modules/odm_meshing/CMakeFiles/odm_meshing.dir/all] Error 2
../../../SuperBuild/install/lib/libopencv_highgui.so.2.4.11: undefined reference to `TIFFIsTiled@LIBTIFF_4.0'
../../../SuperBuild/install/lib/libopencv_highgui.so.2.4.11: undefined reference to `TIFFOpen@LIBTIFF_4.0'
../../../SuperBuild/install/lib/libpcl_io.so: undefined reference to `png_set_longjmp_fn'
../../../SuperBuild/install/lib/libopencv_highgui.so.2.4.11: undefined reference to `TIFFReadEncodedStrip@LIBTIFF_4.0'
../../../SuperBuild/install/lib/libopencv_highgui.so.2.4.11: undefined reference to `TIFFSetField@LIBTIFF_4.0'
../../../SuperBuild/install/lib/libopencv_highgui.so.2.4.11: undefined reference to `TIFFWriteScanline@LIBTIFF_4.0'
../../../SuperBuild/install/lib/libopencv_highgui.so.2.4.11: undefined reference to `TIFFGetField@LIBTIFF_4.0'
../../../SuperBuild/install/lib/libopencv_highgui.so.2.4.11: undefined reference to `TIFFScanlineSize@LIBTIFF_4.0'
../../../SuperBuild/install/lib/libopencv_highgui.so.2.4.11: undefined reference to `TIFFSetWarningHandler@LIBTIFF_4.0'
../../../SuperBuild/install/lib/libopencv_highgui.so.2.4.11: undefined reference to `TIFFSetErrorHandler@LIBTIFF_4.0'
../../../SuperBuild/install/lib/libopencv_highgui.so.2.4.11: undefined reference to `TIFFReadEncodedTile@LIBTIFF_4.0'
../../../SuperBuild/install/lib/libopencv_highgui.so.2.4.11: undefined reference to `TIFFReadRGBATile@LIBTIFF_4.0'
../../../SuperBuild/install/lib/libopencv_highgui.so.2.4.11: undefined reference to `TIFFClose@LIBTIFF_4.0'
../../../SuperBuild/install/lib/libopencv_highgui.so.2.4.11: undefined reference to `TIFFRGBAImageOK@LIBTIFF_4.0'
../../../SuperBuild/install/lib/libopencv_highgui.so.2.4.11: undefined reference to `TIFFReadRGBAStrip@LIBTIFF_4.0'
collect2: error: ld returned 1 exit status
modules/odm_texturing/CMakeFiles/odm_texturing.dir/build.make:397: recipe for target 'bin/odm_texturing' failed
make[2]: *** [bin/odm_texturing] Error 1
CMakeFiles/Makefile2:335: recipe for target 'modules/odm_texturing/CMakeFiles/odm_texturing.dir/all' failed
make[1]: *** [modules/odm_texturing/CMakeFiles/odm_texturing.dir/all] Error 2
../../../SuperBuild/install/lib/libopencv_highgui.so.2.4.11: undefined reference to `TIFFIsTiled@LIBTIFF_4.0'
../../../SuperBuild/install/lib/libopencv_highgui.so.2.4.11: undefined reference to `TIFFOpen@LIBTIFF_4.0'
../../../SuperBuild/install/lib/libpcl_io.so: undefined reference to `png_set_longjmp_fn'
../../../SuperBuild/install/lib/libopencv_highgui.so.2.4.11: undefined reference to `TIFFReadEncodedStrip@LIBTIFF_4.0'
../../../SuperBuild/install/lib/libopencv_highgui.so.2.4.11: undefined reference to `TIFFSetField@LIBTIFF_4.0'
../../../SuperBuild/install/lib/libopencv_highgui.so.2.4.11: undefined reference to `TIFFWriteScanline@LIBTIFF_4.0'
../../../SuperBuild/install/lib/libopencv_highgui.so.2.4.11: undefined reference to `TIFFGetField@LIBTIFF_4.0'
../../../SuperBuild/install/lib/libopencv_highgui.so.2.4.11: undefined reference to `TIFFScanlineSize@LIBTIFF_4.0'
../../../SuperBuild/install/lib/libopencv_highgui.so.2.4.11: undefined reference to `TIFFSetWarningHandler@LIBTIFF_4.0'
../../../SuperBuild/install/lib/libopencv_highgui.so.2.4.11: undefined reference to `TIFFSetErrorHandler@LIBTIFF_4.0'
../../../SuperBuild/install/lib/libopencv_highgui.so.2.4.11: undefined reference to `TIFFReadEncodedTile@LIBTIFF_4.0'
../../../SuperBuild/install/lib/libopencv_highgui.so.2.4.11: undefined reference to `TIFFReadRGBATile@LIBTIFF_4.0'
../../../SuperBuild/install/lib/libopencv_highgui.so.2.4.11: undefined reference to `TIFFClose@LIBTIFF_4.0'
../../../SuperBuild/install/lib/libopencv_highgui.so.2.4.11: undefined reference to `TIFFRGBAImageOK@LIBTIFF_4.0'
../../../SuperBuild/install/lib/libopencv_highgui.so.2.4.11: undefined reference to `TIFFReadRGBAStrip@LIBTIFF_4.0'
collect2: error: ld returned 1 exit status
modules/odm_georef/CMakeFiles/odm_georef.dir/build.make:425: recipe for target 'bin/odm_georef' failed
make[2]: *** [bin/odm_georef] Error 1
CMakeFiles/Makefile2:170: recipe for target 'modules/odm_georef/CMakeFiles/odm_georef.dir/all' failed
make[1]: *** [modules/odm_georef/CMakeFiles/odm_georef.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
Configuration Finished

parang17, 2018-03-23:
@yjmenezes 

- I just did. 
sudo gpasswd -a $USER docker
source ~/.bashrc

- and I checked wheter USER is properly added or not.
docker ps 

- I tried to pull the image accoding to the instruction (https://github.com/OpenDroneMap/OpenDroneMap/wiki/Docker)

docker run -it --rm -v "$(pwd)/images:/code/images" -v "$(pwd)/odm_orthophoto:/code/odm_orthophoto" -v "$(pwd)/odm_georeferencing:/code/odm_georeferencing" opendronemap/opendronemap --mesh-size 100000

But I got the following error message

[image]

Can you help me fixing this issue? Thank you.

yjmenezes, 2018-03-26:
@parang17, 

have you had success running ODM ?

If not, i will try answer you in a more detailed way.

parang17, 2018-03-26:
@yjmenezes 
You are amazing. Thanks for your answer. I haven't tried it, yet. I will try it today and I will let you know the result.

Thank you so much.
Youngjun

parang17, 2018-03-26:
@yjmenezes 
I have tried to run the example case in the test/testdata/images, but I got the following error message. Could you help me fixing this problem? Thank you so much.

[image]

parang17, 2018-03-27:
@yjmenezes
I moved all the files to the other folder and ran again. Looks like it works. I got two new folder (odd_orthophoto and odd_texturing). 

I have one question. How I can get ode_meshing and odd_georeferencing result? 

Thank you so much.

yjmenezes, 2018-03-27:
@parang17 
@parang17,

Nice to hear you run ODM under docker.

You can map any host directory path to  Docker  code/path

take a look here:

https://github.com/yjmenezes/OpenDroneMap/blob/master/odm_docker_readme.txt

good luck

closed 2020-04-14 by smathermather