OpenDroneMap/ODM #792: WebODM error returned child 139

silverefendy, 2018-03-26:
Help me, i got this error.  When first attempt on merging all the image got no error.
With --min-num-features:18000
But the result got cut a little where i need to see it.
So trying to raising the --min-num-features to 24000
After that got this result.
Btw already merging 2 datasets around same quantity.

[INFO]    Running ODM Texturing Cell - Finished
[INFO]    Running ODM Georeferencing Cell
[INFO]    Running georeferencing with OpenSfM transformation matrix
[DEBUG]   running /code/build/bin/odm_georef -bundle
File /var/www/data/ad0c3cf9-2a83-4048-8987-8953a1732b94/opensfm/bundle_r000.out -inputTransform
File /var/www/data/ad0c3cf9-2a83-4048-8987-8953a1732b94/opensfm/geocoords_transformation.txt -input
File /var/www/data/ad0c3cf9-2a83-4048-8987-8953a1732b94/odm_texturing_25d/odm_textured_model.obj -output
File /var/www/data/ad0c3cf9-2a83-4048-8987-8953a1732b94/odm_texturing_25d/odm_textured_model_geo.obj -inputPointCloud
File /var/www/data/ad0c3cf9-2a83-4048-8987-8953a1732b94/opensfm/reconstruction.ply -outputPointCloud
File /var/www/data/ad0c3cf9-2a83-4048-8987-8953a1732b94/odm_25dgeoreferencing/odm_georeferenced_model.ply  -log
File /var/www/data/ad0c3cf9-2a83-4048-8987-8953a1732b94/odm_25dgeoreferencing/odm_georeferencing_log.txt -outputTransform
File /var/www/data/ad0c3cf9-2a83-4048-8987-8953a1732b94/odm_25dgeoreferencing/odm_georeferencing_transform.txt -georef
FileOutputPath /var/www/data/ad0c3cf9-2a83-4048-8987-8953a1732b94/odm_25dgeoreferencing/odm_georeferencing_model_geo.txt
[DEBUG]   running /code/SuperBuild/build/pdal/bin/pdal pipeline -i /var/www/data/ad0c3cf9-2a83-4048-8987-
8953a1732b94/odm_georeferencing/las.json --readers.ply.filename=/var/www/data/ad0c3cf9-2a83-4048-8987-
8953a1732b94/odm_georeferencing/odm_georeferenced_model.ply --writers.las.filename=/var/www/data/ad0c3cf9-2a83-4048-8987-
8953a1732b94/odm_georeferencing/odm_georeferenced_model.las
Segmentation fault (core dumped)
Traceback (most recent call 
last):
  File "/code/run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/odm_georeferencing.py", line 142, in process
    tree.odm_georeferencing_las_json)
  File "/code/opendm/types.py", line 264, in convert_to_las
    '--writers.las.filename={f_out}'.format(**kwargs))
  File "/code/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 139

Got arount 700++ photo

ram 16GB and swap 32GB
Link to data set:
https://drive.google.com/open?id=1oOkYL5bBBncxEYTlLW_N3-cbTQYJSNO_

And thanks for this awesome software.

dakotabenjamin, 2018-03-26:
The dataset leads to a 404 :cry:

silverefendy, 2018-03-26:
Sorry, already make it on again.
Can google drive make it off by default?

dakotabenjamin, 2018-03-27:
I've got it now, thanks

silverefendy, 2018-03-28:
No, i thank you

dakotabenjamin, 2018-03-28:
This issue was moved to http://community.opendronemap.org/t/webodm-error-returned-child-139/381

dakotabenjamin, 2018-03-28:
I'm running the process now

pierotofy, 2018-04-09:
I think this is solved with #803

silverefendy, 2018-04-12:
I'll try merging the photos now.
Without fast ortgophoto being check.

silverefendy, 2018-04-20:
Sorry late for reply.
When i try without fast orthophoto.
It can merge successfully.
Thanks.

closed 2020-01-19 by smathermather