OpenDroneMap/ODM #795: GCP does not seem to be used for georeference

tmizu23, 2018-03-27:
Evev if I run with "--gcp gcp_list.txt", gcp doesn't seem to be used for georeference.
It was written "Missing input in order to use GCP for georeferencing. Using EXIF data instead." in odm_georeferencing_log.txt. and the orthophoto result was also incorrect. 

gcpFileSpecified  at [Georef.cpp#L514](https://github.com/OpenDroneMap/OpenDroneMap/blob/master/modules/odm_georef/src/Georef.cpp#L514) is always false,cause of [odm_georeferencing.py#125](https://github.com/OpenDroneMap/OpenDroneMap/blob/master/scripts/odm_georeferencing.py#L125) is no "-gcpFile {gcp}" option

dakotabenjamin, 2018-03-28:
What version are you running? Can you paste the content of your gcp file?

tmizu23, 2018-03-28:
Sorry it was my misunderstanding.
I update docker image, then fix it.
thank you.

tmizu23, 2018-03-29:
Today I update latest docker image, then this probrem ocuur again.
In some cases, the stable version is uploaded in docker hub, and in some cases, it seems that there are buggy version uploaded.
From 5 days ago until yesterday, it seems there is stable version in docker hub .
I check --help, then buggy version show "--proj" option and stable version don't show it.
Could you upload stable version  in docker hub?

tmizu23, 2018-04-02:
@yjmenezes  thank you

I tried your option and chnage gcp line, but not changed
my running command is this, and attached the log file

docker run -it --rm
-v ${odmdata}/images:/code/images
-v ${odmdata}/opensfm:/code/opensfm
-v ${odmdata}/odm_meshing:/code/odm_meshing
-v ${odmdata}/odm_texturing:/code/odm_texturing
-v ${odmdata}/odm_georeferencing:/code/odm_georeferencing
-v ${odmdata}/odm_dem:/code/odm_dem
-v ${odmdata}/odm_orthophoto:/code/odm_orthophoto
-v ${odmdata}/gcp_list.txt:/code/gcp_list.txt
opendronemap/opendronemap
--resize-to 2048
--min-num-features 8000
--opensfm-process 2
--mesh-size 100000
--gcp gcp_list.txt
--dsm
--dem-resolution 0.1
--dem-terrain-type FlatForest
--orthophoto-resolution 20

[odm_orthophoto_log.txt](https://github.com/OpenDroneMap/OpenDroneMap/files/1867024/odm_orthophoto_log.txt)
[odm_georeferencing_log.txt](https://github.com/OpenDroneMap/OpenDroneMap/files/1867025/odm_georeferencing_log.txt)

yjmenezes, 2018-04-03:
@tmizu23 

you are right, we have a trouble with ODM 0.3.1, it is jumping gcp_list.txt.
Your log shows: Missing input in order to use GCP for georeferencing. Using EXIF data instead.

I will try to figure out the reason.

0x906, 2018-04-20:
is there any update on this issue?

yjmenezes, 2018-05-02:
I run ODM 0.3.1 commit 7816d59  under docker and it is OK.
I run the supplied dataset 
cd  tests/testdata/

closed 2020-04-14 by smathermather