OpenDroneMap/ODM #797: odm_georeferenced_model.ply is not referenced correctly

dakotabenjamin, 2018-03-29:
### What is the problem?

the georeferenced PLY is not georeferenced properly. 

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

The coordinates need to have the origin coordinates in coords.txt added to each value. Then it will be properly referenced. 

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

1. Run ODM on any test dataset
2. Load odm_georeferencing/odm_georeferenced_model.ply and odm_georeferenced_model.las 
3. Check coords.txt for origin coordinates (second line)

pierotofy, 2018-03-29:
Mm, are you sure?

The numbers from coords.txt are simply the 4th column in `odm_georeferencing_transform.txt` (the translation).

These get added here: https://github.com/OpenDroneMap/OpenDroneMap/blob/master/modules/odm_georef/src/Georef.cpp#L1409

The changes are very recent (we just merged a PR for it).

dakotabenjamin, 2018-03-29:
@smathermather can you rerun with latest ODM or WebODM and check?

closed 2020-01-19 by smathermather