OpenDroneMap/ODM #798: PDAL: writers.las: Unable to convert scaled value (1.564621925e+25) to int32 for dimension 'Z'

gmhuili, 2018-03-30:
****************************************
PLEASE REMOVE THIS NOTE AFTER READING IT!

First of all, thank you for taking the time to report an issue.

Before you continue, make sure you are in the right place. Please open an issue only to report faults and bugs. For questions and discussion please open a topic on http://community.opendronemap.org/c/opendronemap. 

Please use the format below to report bugs and faults. It will help improve the resolution process. 
****************************************

### How did you install OpenDroneMap? (Docker, natively, ...)?
Docker.

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Firefox 59 on Ubuntu Linux

### What is the problem?

Building objmodel:
	Saving model... done.
Whole texturing procedure took: 1427.13s
[INFO]    Running ODM Texturing Cell - Finished
[INFO]    Running ODM Georeferencing Cell
[INFO]    Running georeferencing with OpenSfM transformation matrix
[DEBUG]   running /code/build/bin/odm_georef -bundleFile /code/opensfm/bundle_r000.out -inputTransformFile /code/opensfm/geocoords_transformation.txt -inputFile /code/odm_texturing/odm_textured_model.obj -outputFile /code/odm_texturing/odm_textured_model_geo.obj -inputPointCloudFile /code/opensfm/depthmaps/merged.ply -outputPointCloudFile /code/odm_georeferencing/odm_georeferenced_model.ply  -logFile /code/odm_georeferencing/odm_georeferencing_log.txt -outputTransformFile /code/odm_georeferencing/odm_georeferencing_transform.txt -georefFileOutputPath /code/odm_georeferencing/odm_georeferencing_model_geo.txt
[DEBUG]   running /code/SuperBuild/build/pdal/bin/pdal pipeline -i /code/odm_georeferencing/las.json --readers.ply.filename=/code/odm_georeferencing/odm_georeferenced_model.ply --writers.las.filename=/code/odm_georeferencing/odm_georeferenced_model.las
PDAL: writers.las: Unable to convert scaled value (1.564621925e+25) to int32 for dimension 'Z' when writing LAS/LAZ file /code/odm_georeferencing/odm_georeferenced_model.las.

Traceback (most recent call last):
  File "/code/run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/odm_georeferencing.py", line 142, in process
    tree.odm_georeferencing_las_json)
  File "/code/opendm/types.py", line 264, in convert_to_las
    '--writers.las.filename={f_out}'.format(**kwargs))
  File "/code/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

No this kind of issue.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

It happens every time and could be caused by the pictures.  

Build:
docker build -t my_odm_image .

Run:
docker run -it --rm -v $(pwd)/gcp_list.txt:/code/gcp_list.txt -v $(pwd)/images:/code/images -v $(pwd)/opensfm:/code/opensfm -v $(pwd)/odm_meshing:/code/odm_meshing -v $(pwd)/odm_georeferencing:/code/odm_georeferencing -v $(pwd)/odm_orthophoto:/code/odm_orthophoto -v $(pwd)/odm_texturing:/code/odm_texturing my_odm_image  --mesh-size 100000

pierotofy, 2018-04-11:
Related: http://community.opendronemap.org/t/copr-sample-dataset-failing/446/8

Can be reproduced by using this dataset, using `gcp_list.txt` for GCPs. https://github.com/OpenDroneMap/odm_data_copr

The problem is (probably) caused by `odm_georef`. The final transform (odm_georeferencing_transform.txt) is way out.

```
[ 0.00000000,	0.00000000,	0.00000000,	1000000000000000019884624838656.00000000 ]
[ 0.00000000,	0.00000000,	0.00000000,	1000000000000000019884624838656.00000000 ]
[ 0.00000008,	0.00000008,	1.00000000,	0.00000000 ]
[ 0.00000000,	0.00000000,	0.00000000,	1.00000000 ]
```

In turn this messes up `odm_georeferencing.py`, which calls pdal.

Help troubleshooting this is welcome.

adamsteer, 2018-04-21:
Hi @pierotofy, @gmhuili - for:

`PDAL: writers.las: Unable to convert scaled value (1.564621925e+25) to int32 for dimension 'Z'...`

...PDAL is trying to stuff that value into int32 because that is the ASPRS spec for storing the dimension. You're correct about the matrix - if that transformation is correct, and assuming I'm reading it correctly, one workaround is adding an offset when writing out to LAS/LAZ - for example as a part of a PDAL pipeline:

```
{
"type": "writers.las",
"offset_x":"10000000000000000198846",
"offset_y": "10000000000000000198846",
...
...

}
```
...where offsets could be inferred as required? Unfortunately I don't know the ODM codebase, so can't test this idea - whip you up a pull request :/

gmhuili, 2018-04-23:
Thanks @adamsteer!  Will try it again once I get a chance.

adamsteer, 2018-04-23:
Hi @gmhuili - without some work inside ODM, or changing the coordinates for your georeferencing, it'll likley still break right now.

Unfortunately I haven't seen where PDAL is invoked - `https://github.com/OpenDroneMap/OpenDroneMap/blob/master/modules/odm_georef/src/Georef.cpp` is where the last message output lives, but I can't yet see where it's writing pipeline is created or how to implement something like automagically creating an appropriate offset.

jtinguely, 2018-07-03:
+1. I see this error quite often as well, I am running docker on windows 10 pro machine. I install ODM via Dockerhub.

Rob-Besaans, 2018-07-04:
+1. Same issue as @jtinguely. Also windows 10 pro 32gb machine SSD. I installed docker from the WebODM Windows Commercial install package. I am testing this with the ODM NIR sample dataset however.

https://github.com/OpenDroneMap/odm_data_rv_nir

pierotofy, 2018-09-29:
Hey @Rob-Besaans, @adamsteer, @gmhuili, are you still having this issue with the latest version of ODM? I'm having a hard time replicating this since we've released 0.4. Could you try to reprocess your datasets and let me know if the issue is still there? The opensfm pipeline has been updated with 0.4, so this could have fixed the problem.

hy9be, 2018-10-09:
@pierotofy I can still reproduce this issue on 0.4RC1. Use the first 50 images of KITTI sequence 00 dataset: https://drive.google.com/file/d/0ByaBRAPfmgEqdXhJRmktQ2lsMEE/view

Rob-Besaans, 2018-10-09:
Hi guys. Sorry I have a couple of business deadlines to meet over the next
week or two. Please keep me in the loop. I'm happy to test further after my
works finished.

Cheers

Robert Besaans
Business Analyst

skorasaurus, 2019-08-16:
I've experienced this well on my data set from and odm docker image was built around (around 2019-08-05 or so.). 

full logs and information is at https://community.opendronemap.org/t/unable-to-convert-scaled-value-1e-32-to-int32-pdal-with-grayscale-colorspace-images/2665

closed 2020-04-14 by smathermather