OpenDroneMap/ODM #799: Using OpenDroneMap without geo data

Harvie, 2018-03-30:
Hello,
is there way to reconstruct point cloud / mesh from photos usinf ODM without having the gps coords? I don't need absolute position, just get 3D model for manual editing and aligning or 3d printing.

Right now i get this:

```
[WARNING] Could not generate coordinates file. Ignore if there is a GCP file. Error: Error in OdmExtractUtm:
Image is missing GPS Latitude data
For more detailed information, see log file.

[ERROR]   Could not find file /code/odm_georeferencing/coords.txt
```

I just want ODM to ignore this and give me output mesh with relative coords. I don't have gps in txt, gcp or exif. i just don't want to use them at all. I know this is possible with colmap. But i like ODM more.

Thanks for your time.

pierotofy, 2018-03-31:
Could you copy/paste the full output of processing your task?

Harvie, 2018-03-31:
I've used this command:

`docker run -it --rm -v $(pwd)/images:/code/images opendronemap/opendronemap --mesh-size 100000 --force-ccd 1  --help`

I don't have it set up right now. But i think there was nothing else unusual. Just info about loading the JPGs and then this.

pierotofy, 2018-04-11:
Related: http://community.opendronemap.org/t/could-not-find-file-odm-georeferencing-coords-txt/487

closed 2018-04-20 by pierotofy

smathermather, 2018-04-20:
@Harvie -- this is fixed and tested on master. Give it a spin when you get a chance!

PR #806 (2018-04-20): Fixes to get non-georeferenced images to process correctly
A recent update broke the ability to process images that have no EXIF GPS information or GCPs datasets. These changes allow users to process non-georeferenced images again.

Fixes #799, http://community.opendronemap.org/t/could-not-find-file-odm-georeferencing-coords-txt/487