OpenDroneMap/ODM #801: Global color adjustment problem on mosaic

giovannicimolin, 2018-04-06:
### How did you install OpenDroneMap? (Docker, natively, ...)?

Docker

### What's your browser and operating system? 

Ubuntu 14.04 - Running on AWS EC2 r3.4xlarge (16xCPU, 120GB-RAM, 320GB-SSD)

### What is the problem?

The generated orthophotos have a gradient of colors going from very white on one side of the mosaic to very dark on another side. This has been observed in many cases, shown on the links below:

https://drive.google.com/open?id=1Viv-HkDU2n1ZMcU6p7Zaws-lCupZFM21
https://drive.google.com/open?id=19AnUksletBzzS1JbpzigVDHKJrObGv8_
https://drive.google.com/open?id=1RIIORr52a8FkXwG-gLTPu5b1a2Ln8r6H
https://drive.google.com/open?id=1x46MxAh9m7v9wa5Z43X_8iEVc1f7bwM-

Processing these datasets using the following parameters:
```javascript
{
            "use-hybrid-bundle-adjustment": true,
            "min-num-features": 12000,
            "mesh-octree-depth": 6,
            "orthophoto-resolution": 20
}
```

Using `--texturing-skip-global-seam-leveling`, the problem doesn't happen, however, the result is not as normally expected:
https://drive.google.com/open?id=1hYRNdbVXmCXbBEFz93dvY2kKity_Q_3W

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

An orthomosaic with good color balance.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)
I'm working on providing a dataset. The parameters used were:

```javascript
{
            "use-hybrid-bundle-adjustment": true,
            "min-num-features": 12000,
            "mesh-octree-depth": 6,
            "orthophoto-resolution": 20
}
```
I've also made tests without using `--mesh-octree-depth`.

giovannicimolin, 2018-04-06:
If you can point me to a possible parameter mistake or lines of code to look at I'd be happy to help. :smile:

pierotofy, 2018-04-09:
Do you get better results if you turn on `--texturing-tone-mapping gamma`?

pierotofy, 2018-04-09:
Also, screenshots are great, would it be possible to have access to the original images as well?

smathermather, 2018-04-10:
@giovannicimolin -- FYI, you don't have to make the images open to share them. If you want to share them with @pierotofy, he can probably agree to not share them further.

farmdude, 2018-04-11:
Thanks for pointing me here @pierotofy 

First image is --texturing-tone-mapping none
[image]
Second image is --texturing-tone-mapping gamma
[image]

However, the problem, in my opinion, becomes more pronounced when you turn gamma on.

giovannicimolin, 2018-04-11:
Here is one of the mosaics with ` --texturing-tone-mapping gamma`:
[image]

pierotofy, 2018-04-19:
Possible fix should be available as soon as https://github.com/OpenDroneMap/mvs-texturing/pull/3 is merged.

foxsake, 2018-04-26:
I had the same issue. seen on the first pic
[image]
the update fixed the issue.
[image]

-Thanks   @pierotofy!

pierotofy, 2018-04-26:
Thanks for testing @foxsake !

farmdude, 2018-04-26:
@pierotofy your fix worked for me! Thanks!

giovannicimolin, 2018-04-29:
@pierotofy The solves all the problems! Thank you!

closed 2018-04-29 by giovannicimolin