OpenDroneMap/ODM #804: [Idea] odm.exe wrapper around WSL on Windows

pierotofy, 2018-04-13:
Started looking a little more into WSL under Windows 10: https://en.wikipedia.org/wiki/Windows_Subsystem_for_Linux

After having successfully compiled and run ODM using WSL, I was wondering how to improve things. Docker's great, but it's not the easiest it could be. Ideally I would like to run something like:

```
C:\ odm.exe --project-path Users\myuser\drone myproject
```

And be done with it. No compilation should be required and paths should be expressed using Windows standards.

WSL gets us half-way there. You can't just make a tarball of ODM files after compilation, because lib dependencies are embedded within the Ubuntu system, so we still need the full Ubuntu system. But you cannot release a full distro on the Windows Store (not at the moment, anyway).

An interesting module/subproject for ODM would be the creation of a `odm.exe` wrapper that:
 * Checks that Ubuntu is installed with WSL
 * Downloads a tarball of the docker image's filesystem (the entire system) if necessary
 * chroots into the above filesystem
 * Converts Windows style paths and locations to work in the Ubuntu WSL
 * Executes `./run.sh`.

Seems doable to me. Just need to find time 😄 or other volunteers.

MattWenham, 2019-04-15:
Seems you're about a year ahead of me, but I did just get ODM to run successfully on a very low-powered tablet (I'm on vacation) with 6GB of RAM and 4GB of virtual memory.

The above sounds like a great proposition, I will investigate further if I find the time.

Another alternative might be a PowerShell script?

pierotofy, 2019-04-15:
PowerShell could work also, yes.

MattWenham, 2019-04-15:
I don't think it even needs PowerShell, a straight batch file might work.

If I'm tarballing the `OpenDroneMap_v0_3_1` folder, which file extensions can I safely exclude for the software to work when restored? I'm thinking of source and object files?

smathermather, 2020-04-14:
@pierotofy -- I will close for now.

closed 2020-04-14 by smathermather

pierotofy, 2020-04-14:
Yep, the idea was good in principle, but MS has back-tracked on WSL 1.0 and their implementation of WSL 2.0 is a half-baked solution IMO. Pity...

smathermather, 2020-04-15:
That's such a shame.