OpenDroneMap/ODM #810: Integrate newer OpenSfM

smathermather, 2018-04-24:
Latest version of OpenSfM does local/global bundle adjustment in a well tuned way by default. We should pull in some of these late changes and test, as it should be much more performant doing bundle adjustment on large datasets out of the box, as it were.

(fyi, @dakotabenjamin @pierotofy)

PeterSprague, 2018-06-12:
Using master now on both a 64 core workstation and a 4 core Atom SBC.  No errors with smaller sets of images.  Subjectively seems to have dropped processing time from 1:15 --> 0:52 minutes on my 119 image test set.

Want help for tests?

smathermather, 2018-06-12:
Tests and a pull request would be most appreciated!

PeterSprague, 2018-06-12:
I can only test for Docker environment, plus I use debian stretch as my base image, a custom docker multi-stage build, along with master or latest releases for source builds.

Are there a set of defined tests to run?

smathermather, 2018-06-13:
No formal set of tests at this time -- it's useful to test the following: does it run? Is there a significant change in speed? Do the output results change much (subjective). If there's a pull request, than Piero, Dakota, or I will run some tests against known datasets.

PeterSprague, 2018-06-13:
does it run? Yes, in my environment
Is there a significant change in speed? Yes, in my environment, see above
Do the output results change much (subjective)? Not really tracked, but could compare if I use the current ODM produced image with and w/o newest OpenSfM.  Been more concerned with error free runs and increased speed for running on lower power equipment including arm64

PR - can do once I set up a stock ODM environment and test

mordka, 2018-11-13:
@PeterSprague any update on this? do you need help in testing? I would like to see OpenSfM v0.2.0 in ODM!

PeterSprague, 2018-11-13:
Nothing new.  Used my fork in our Summer field research season on a 4 core fanless Atom unit.  Had no or very slow upload speeds, so working off-line on a dedicated unit paid dividends.

To speed up analysis, I have moved back to working with individual georeferenced images instead of orthos to feed my Keras ML models to access forest health.  End goal is to have near-real time analysis.  Also were significant issues with lack of determinism in the output. 

My fork uses mostly master for all the source builds dependencies.  Was based off ODM code from later Spring/2018, so didn't have the latest PR that could tighten up the lack of determinism problems.

pierotofy, 2020-01-20:
This can probably be closed.

closed 2020-01-20 by pierotofy

smathermather, 2020-01-20:
But isn't this issue evergreen? :) 🌲

pierotofy, 2020-01-20:
lol, I suppose it could be.