OpenDroneMap/ODM #811: Error when run in the docker

bailingxiao, 2018-04-24:
I use this commands:
docker run -it --rm -v E:\project\OpenDroneMap\odm_test_1/images:/code/images -v E:\project\OpenDroneMap\odm_test_1/odm_texturing:/code/odm_texturing opendronemap/opendronemap
An error occurred "docker: Error response from daemon: Mount denied:
The source path doesn't exist and is not known to Docker."
Then i create a new file called 'odm_texturing',try again,it says"docker: Error response from daemon: 
error while creating mount source path'/host_mnt/e/project/OpenDroneMap/odm_test_1/odm_texturing': mkdir /host_mnt/e: file exists."
What should i do to solve this error?What is the"code/images"?

pierotofy, 2018-04-24:
It might be easier to place your images within your user directory (C:\Users\youruser\), docker on Windows by default only allows paths within a user's home directory to be mounted. See http://support.divio.com/local-development/docker/how-to-use-a-directory-outside-cusers-with-docker-toolbox-on-windowsdocker-for-windows for instructions on how to mount paths outside of your home directory.

bailingxiao, 2018-04-25:
@pierotofy Thank you for your help!
It's really a problem with the path，i have solved it!

closed 2018-04-25 by pierotofy