OpenDroneMap/ODM #818: ODM how to get the ODM_mesh result folder?

vgorte, 2018-05-08:
### How did you install OpenDroneMap? (Docker, natively, ...)?

Installed via Docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/w/APBRDPM

### What is the problem?
I have never used ODM before so I might be missing something obvious here.
How do I get the odm_mesh results? They are listed [here](https://github.com/OpenDroneMap/OpenDroneMap/wiki/Output-and-Results) but the created "odm_mesh" folder is empty. 
I used this terminal-command: 

```
docker run -it --rm \
    -v $(pwd)/images:/code/images \
    -v $(pwd)/opensfm:/code/opensfm \
    -v $(pwd)/odm_meshing:/code/odm_meshing \
    -v $(pwd)/odm_texturing:/code/odm_texturing \
    -v $(pwd)/odm_georeferencing:/code/odm_georeferencing \
    -v $(pwd)/odm_orthophoto:/code/odm_orthophoto \
    -v $(pwd)/odm_mesh:/code/odm_mesh \
    my_odm_image  
```
Any idea why I don't get anything? Everything else works fine.

smathermather, 2018-05-09:
Have you tried ```odm_texturing``` as your directory? This would contain the final textured mesh, which is likely what you are looking for.

vgorte, 2018-05-09:
You are correct. I don't really know why but I managed to first read 
https://github.com/OpenDroneMap/OpenDroneMap/wiki/Output-and-Results
and after posting this question I read the "View results" from the ReadMe.md ...
Is it safe to say that the above link is outdated ? 

Besides that, this issue can be closed. Thank you @smathermather ! 
Have a nice day!

smathermather, 2018-05-09:
Hey @vgorte -- do you mind fixing it in the wiki?

closed 2018-05-09 by smathermather

vgorte, 2018-05-09:
Hey @smathermather  
I fixed it in the wiki.