OpenDroneMap/ODM #823: numpy.linalg.linalg.LinAlgError: SVD did not converge

Timple, 2018-05-18:
### How did you install OpenDroneMap? (Docker, natively, ...)?

Docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Chrome 66 on Linux
(Ubuntu 16.04)

### What is the problem?

numpy.linalg.linalg.LinAlgError: SVD did not converge

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

No error?
Or a more usefull description on what could be the rootcause (too little images? to distorted images? etc)

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Using the dropbox data below:
https://www.dropbox.com/sh/qu46hedvfcz3ox7/AABa-9-RDGzMvhC93bp2scDta?dl=0

Timple, 2018-08-13:
Why was this labeled as enhancement? It does not function at the moment, so I'd rather call it a bug?
Or help-wanted if it was my misdoing.

pierotofy, 2020-09-13:
The latest version of ODM is able to process this (even though with some artifacts):

[image]

So this should be solved.

closed 2020-09-13 by pierotofy

smathermather, 2020-09-13:
Those artifacts are in areas without good coverage, so I would call this resolved. I know it's been a long time @Timple, but if you get a chance to rerun, it should work now.

Timple, 2020-09-14:
This is amazing! I didn't get a 'closed due to inactivity' or something but someone actually took the effort to dig into this problem to see if it was still valid. Thanks for that!

I do not have the same os anymore, but since it runs in docker I assume that won't change the original problem. I'll get to it somewhere in the next two ~~years~~ days :slightly_smiling_face:

Timple, 2020-09-14:
Worked like a charm! I ran with all defaults and had no convergence problems.
As @smathermather mentioned there are obviously artifacts where pictures are missing. 

Interestingly enough a really big tree is completely missing :smile: but that is probably some tuning of the settings. And is unrelated to this problem here, so thank you for coming back on this!
[image]

smathermather, 2020-09-15:
Those trees disappearing is a classic photogrammetry problem. You need to fly quite high or with very high overlap (83%) to successfully reconstruct many trees.

Fantastic. We don't get to all the old bugs, but we fix a lot of stuff, and it's great to get confirmation when we do.