OpenDroneMap/ODM #824: Errors in compute_depthmaps (Exception: Child returned 1)

Dan931, 2018-05-18:
Hello, first of all just want to thank you guys for the excellent work done here. Unfortunately I have encountered some errors that I hope someone can guide me to resolve it. I installed OpenDroneMap natively on Ubuntu 16.04. After installation I tried to run ODM using one of your sample data (caliterra). I ran ODM by typing `./run.sh -i ../odm_data_caliterra/images/ ODMSampleSet` and encountered errors at the running of OpenSfM cell and here is the log: 

[DEBUG]   running PYTHONPATH=/home/Dan/OpenDroneMap/SuperBuild/install/lib/python2.7/dist-packages /home/Dan/OpenDroneMap/SuperBuild/src/opensfm/bin/opensfm compute_depthmaps /home/Dan/Processed/ODMSampleSet/opensfm
2018-05-19 03:08:42,601 INFO: Computing neighbors
2018-05-19 03:09:29,042 INFO: Computing depthmap for image IMG_9399.jpg with PATCH_MAP
2018-05-19 03:09:29,403 INFO: Computing depthmap for image IMG_9420.jpg with PATCH_MAP
2018-05-19 03:09:29,818 INFO: Computing depthmap for image IMG_9375.jpg with PATCH_MAP
2018-05-19 03:09:30,185 ERROR: Exception on child. Arguments: (<opensfm.dataset.DataSet instance at 0x7fb08a956a70>, [<opensfm.types.Shot object at 0x7fb08291fad0>, <opensfm.types.Shot object at 0x7fb08291fd10>, <opensfm.types.Shot object at 0x7fb08291ff50>, <opensfm.types.Shot object at 0x7fb0829211d0>, <opensfm.types.Shot object at 0x7fb082921410>, <opensfm.types.Shot object at 0x7fb082921650>, <opensfm.types.Shot object at 0x7fb082921890>, <opensfm.types.Shot object at 0x7fb082921ad0>, <opensfm.types.Shot object at 0x7fb082921d10>, <opensfm.types.Shot object at 0x7fb082921f50>, <opensfm.types.Shot object at 0x7fb0829281d0>], 60.69756500843941, 76.98564465660867, <opensfm.types.Shot object at 0x7fb08291fad0>)
2018-05-19 03:09:30,185 ERROR: Unknown depthmap method type (must be BRUTE_FORCE, PATCH_MATCH or PATCH_MATCH_SAMPLE)
Traceback (most recent call last):
  File "/home/Dan/OpenDroneMap/SuperBuild/src/opensfm/opensfm/dense.py", line 57, in compute_depthmap_catched
    compute_depthmap(arguments)
  File "/home/Dan/OpenDroneMap/SuperBuild/src/opensfm/opensfm/dense.py", line 105, in compute_depthmap
    'Unknown depthmap method type '
ValueError: Unknown depthmap method type (must be BRUTE_FORCE, PATCH_MATCH or PATCH_MATCH_SAMPLE)
2018-05-19 03:09:30,246 INFO: Computing depthmap for image IMG_9426.jpg with PATCH_MAP
2018-05-19 03:09:30,591 ERROR: Exception on child. Arguments: (<opensfm.dataset.DataSet instance at 0x7fcb83841a70>, [<opensfm.types.Shot object at 0x7fcb7b017ad0>, <opensfm.types.Shot object at 0x7fcb7b017d10>, <opensfm.types.Shot object at 0x7fcb7b017f50>, <opensfm.types.Shot object at 0x7fcb7b0191d0>, <opensfm.types.Shot object at 0x7fcb7b019410>, <opensfm.types.Shot object at 0x7fcb7b019650>, <opensfm.types.Shot object at 0x7fcb7b019890>, <opensfm.types.Shot object at 0x7fcb7b019ad0>, <opensfm.types.Shot object at 0x7fcb7b019d10>, <opensfm.types.Shot object at 0x7fcb7b019f50>, <opensfm.types.Shot object at 0x7fcb7b0201d0>], 59.49645211246254, 82.81069111829342, <opensfm.types.Shot object at 0x7fcb7b017ad0>)
2018-05-19 03:09:30,591 ERROR: Unknown depthmap method type (must be BRUTE_FORCE, PATCH_MATCH or PATCH_MATCH_SAMPLE)
Traceback (most recent call last):
  File "/home/Dan/OpenDroneMap/SuperBuild/src/opensfm/opensfm/dense.py", line 57, in compute_depthmap_catched
    compute_depthmap(arguments)
  File "/home/Dan/OpenDroneMap/SuperBuild/src/opensfm/opensfm/dense.py", line 105, in compute_depthmap
    'Unknown depthmap method type '
ValueError: Unknown depthmap method type (must be BRUTE_FORCE, PATCH_MATCH or PATCH_MATCH_SAMPLE)
2018-05-19 03:09:48,246 INFO: Cleaning depthmap for image IMG_9399.jpg
Traceback (most recent call last):
  File "/home/Dan/OpenDroneMap/SuperBuild/src/opensfm/bin/opensfm", line 34, in <module>
    command.run(args)
  File "/home/Dan/OpenDroneMap/SuperBuild/src/opensfm/opensfm/commands/compute_depthmaps.py", line 25, in run
    dense.compute_depthmaps(data, graph, reconstructions[0])
  File "/home/Dan/OpenDroneMap/SuperBuild/src/opensfm/opensfm/dense.py", line 43, in compute_depthmaps
    parallel_map(clean_depthmap_catched, arguments, processes)
  File "/home/Dan/OpenDroneMap/SuperBuild/src/opensfm/opensfm/context.py", line 38, in parallel_map
    return list(e.map(func, args))
  File "/usr/local/lib/python2.7/dist-packages/loky/process_executor.py", line 719, in _chain_from_iterable_of_lists
    for element in iterable:
  File "/usr/local/lib/python2.7/dist-packages/loky/_base.py", line 584, in result_iterator
    yield future.result()
  File "/usr/local/lib/python2.7/dist-packages/loky/_base.py", line 431, in result
    return self.__get_result()
  File "/usr/local/lib/python2.7/dist-packages/loky/_base.py", line 382, in __get_result
    raise self._exception
loky.process_executor.BrokenProcessPool: A process in the executor was terminated abruptly while the future was running or pending.
Traceback (most recent call last):
  File "/home/Dan/OpenDroneMap/run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/home/Dan/OpenDroneMap/scripts/run_opensfm.py", line 164, in process
    (context.pyopencv_path, context.opensfm_path, tree.opensfm))
  File "/home/Dan/OpenDroneMap/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1

I have tried changing the number of processes but I still face the same issues and I cannot find anyone that have similar problems. Installation of ODM was fine as there were no errors. Appreciate it if someone can help me with my problems.

Here is the full log:
[logfile.txt](https://github.com/OpenDroneMap/OpenDroneMap/files/2018536/logfile.txt)

dakotabenjamin, 2018-05-18:
This looks like a bug introduced by the [latest PR](https://github.com/OpenDroneMap/OpenDroneMap/pull/822/files). We'll check it out, in the meantime can you test the different `--opensfm-depthmap-options` options and see if that works?

closed 2018-05-18 by pierotofy

dakotabenjamin, 2018-05-18:
@dan931 thanks for pointing this out, it's fixed now.

PR #825 (2018-05-18): Change PATCH_MAP to PATCH_MATCH in config
@pierotofy fixes #824