OpenDroneMap/ODM #828: OpenSfM does not handle EPSG codes

pierotofy, 2018-05-24:
https://github.com/mapillary/OpenSfM/blob/master/opensfm/io.py#L401

If you use any EPSG code definition (ex. "EPSG:4326") to specify a GCP projection, OpenDroneMap handles it OK, but is not valid in OpenSfM (see io.py).

Options:

1) We add support for EPSG codes in OpenSfM.
2) We remove support for EPSG codes in ODM.

Thoughts?

smathermather, 2018-05-24:
Related, but GCPi doesn't support EPSG codes either, because it's built on a port of proj4 rather than gdal.

pierotofy, 2018-05-24:
So... should we just remove EPSG codes and just use proj4 / UTM strings?

smathermather, 2018-05-24:
Probably. I think otherwise we either need to deal with them ourselves, or introduce a dependency to OpenSfM.

smathermather, 2018-05-24:
Since we have gdal as a dependency, we might be able to use gdal to convert EPSG codes to proj4 strings. I don't know enough about it to say for sure offhand.

pierotofy, 2018-05-24:
That could work.

`gdalsrsinfo epsg:4326 -o proj4`
'+proj=longlat +datum=WGS84 +no_defs '

We would need to create a new GCP file to feed to OpenSfM.

dakotabenjamin, 2018-05-24:
I'd like to advocate for epsg codes as a far more accessible block of text than proj4 strings

pierotofy, 2018-05-24:
Seeing that calling gdalsrsinfo is straightforward, I'm for keeping EPSG codes also.

smathermather, 2018-05-24:
Sounds like a plan.

pierotofy, 2018-08-28:
Partecipants at FOSS4G have reported this as being an important issue for them.

smathermather, 2018-08-29:
👍

pierotofy, 2019-06-21:
This is done in the `speedimprov` branch.

closed 2019-06-21 by pierotofy