OpenDroneMap/ODM #833: odm_georef needs better info when referencing non-existent mesh

smathermather, 2018-06-05:
### How did you install OpenDroneMap? (Docker, natively, ...)?

Natively

### What is the problem?

If odm_georef is run referencing a mesh that doesn't exist, rather than indicating the mesh doesn't exist, it states the following:

```
Error in Georef:
Problem reading header in modelfile!
```

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

It should state the the file doesn't exist.

### How can we reproduce this?

Modify the odm_georef command from a successful run to point to a non-existent input mesh.

pierotofy, 2020-09-11:
Does this problem still occur?

smathermather, 2020-09-12:
This has been fixed at least since you put bananas on the map. https://github.com/OpenDroneMap/ODM/pull/1068

closed 2020-09-12 by smathermather