OpenDroneMap/ODM #834: Possible to install Natively on Ubuntu 18.04 LTS?

Dan931, 2018-06-06:
Hello I was just wondering if it is currently possible to install natively in Ubuntu 18.04 LTS? If it is possible there are any edits I should make? I tried installing as in Ubuntu 16.04, (it is working in my other computer using ubuntu 16.04) but I am faced with the error for 18.04: 

Make Error at modules/odm_georef/CMakeLists.txt:18 (find_package):
  Could not find a package configuration file provided by "OpenCV" with any
  of the following names:

    OpenCVConfig.cmake
    opencv-config.cmake

  Add the installation prefix of "OpenCV" to CMAKE_PREFIX_PATH or set
  "OpenCV_DIR" to a directory containing one of the above files.  If "OpenCV"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!

Using docker to install ODM will be my last resort. Any guidance is appreciated thank you!

pierotofy, 2018-06-18:
Hey @Dan931, it should be definitely possible to install ODM on Ubuntu 18.04, but we currently target 16.04 so we haven't done much testing. Some packages (like OpenCV) might need to be configured or updated to work on 18.04. I couldn't tell exactly how since I haven't tried to install it on 18.04.

pierotofy, 2019-01-07:
The end of life for Ubuntu 16.04 is April 2021. Even though it's 3 years away, we should start thinking about upgrading at some point in the near future, as many users are already looking at 18.04 as the default LTS Ubuntu version.

smathermather, 2019-01-07:
Agreed.

suricactus, 2019-09-09:
ping

omaralvarez, 2019-11-15:
pong

smathermather, 2019-11-16:
Related:
https://community.opendronemap.org/t/native-install-on-ubuntu-18-04/3151

pierotofy, 2020-10-22:
This has been done :clinking_glasses:

closed 2020-10-22 by pierotofy