OpenDroneMap/ODM #837: Replace odm_meshing with PoissonRecon

pierotofy, 2018-06-07:
https://github.com/mkazhdan/PoissonRecon

The PoissonRecon program is faster than PCL's implementation and includes the new Screened Poisson Reconstruction code.

http://www.cs.jhu.edu/~misha/Code/PoissonRecon/Version10.01/

We should expose/adapt the following parameters:

[--depth <maximum reconstruction depth>=8] (this is --mesh-octree-depth)
[--pointWeight <interpolation weight>=2.000e+00 * <b-spline degree>] (This floating point value specifies the importance that interpolation of the point samples is given in the formulation of the screened Poisson equation.
The results of the original (unscreened) Poisson Reconstruction can be obtained by setting this value to 0. Default is 4 )

The --linearFit and --normals parameters should always be passed.

--mesh-octree-divider parameter needs to be removed.

hy9be, 2018-07-08:
@pierotofy I saw all items in ODM output improvements project were moved to Done: https://github.com/OpenDroneMap/OpenDroneMap/projects/2

On which branch should I get the code with all these changes? I would like to test it out. Thanks!

pierotofy, 2018-07-08:
Hey @hy9be, some changes are in the `odm_output_improvements` branch, but not all of them. We will post an announcement when things are merged into `master`. Hopefully very soon :+1: 

cc @smathermather

hy9be, 2018-07-09:
Thanks! Look forward to it! @pierotofy @smathermather

pierotofy, 2018-09-16:
Done!

closed 2018-09-16 by pierotofy