OpenDroneMap/ODM #841: No output when running ODM in docker under -d (--detach) mode

dodge-ttu, 2018-06-07:
To begin with, thank you so much to the originators for creating ODM. I am blown away by this software and extremely grateful.

I am running ODM on a DO Ubuntu 16.04 server in a docker conatiner. It's easy and works great.

My problem occurs when I run docker in detach mode. I never get any output. I have tried on small sets that I know should process and I get nothing, in terms of output, except the creation of the target directories. I need to run docker this way so I can close my terminal and have the image processing continue.

Does anyone have an idea why running ODM in a detached docker might not produce any output?

Thanks!

pierotofy, 2018-06-08:
Hey @dodge-ttu, that's normal, you are running in detached mode.

Take a look at `screen` to run programs in a shell and being able to close it and continue processing.

https://uisapp2.iu.edu/confluence-prd/pages/viewpage.action?pageId=115540034

dodge-ttu, 2018-06-11:
Ahhh.... That is exactly what I needed. Thank you!

closed 2018-06-11 by pierotofy