OpenDroneMap/ODM #845: Add compressed LAZ files to output as alternative to LAS

smathermather, 2018-06-15:
### What is the problem?

Current PDAL LAS output is uncompressed.

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

We should add compression to output.

#### Tasks:
* Change the output file name to LAZ
* Set the compression to true as per https://pdal.io/stages/writers.las.html.

These can be changed here: https://github.com/OpenDroneMap/OpenDroneMap/blob/master/opendm/types.py#L244

As long as PDAL is compiled with laszip support, that should be all that needs done. If that doesn't work, we'll need to change how we are building PDAL, but as PDAL has made lots of progress on the packaging side of things, this might be worth an upgrade anyway.

pierotofy, 2018-06-16:
Yeah it should be fairly easy.

jgrocha, 2018-06-16:
I've changed the `pdal` pipeline to write the compressed `*.laz` file, but as @smathermather  suspected, `pdal` is not being compiled with `LASzip` support.

The error I get is:
```
PDAL: writers.las: Can't write LAZ output.  PDAL not built with LASzip or LAZperf.
```
Looking at the code https://github.com/OpenDroneMap/OpenDroneMap/blob/f09b42ec81bdfcbf2cbc5d5050e3bda0f95e43e0/SuperBuild/cmake/External-PDAL.cmake#L36 the LASZIP is included in the PDAL building process, but that is probably failing.

We need to compile LASzip before installing PDAL, right? Am I able to do that? Hum, I'll try. Help is appreciated :-)

pierotofy, 2018-06-16:
Yeah LASzip should be added as a depedency, just like `hexer`.

https://github.com/LASzip/LASzip

See the `External-Hexer.cmake` file, create one for LASzip, then update `External-PDAL.cmake` to include LASzip as a dependency (on line 5), finally add a line to include LASzip via `SETUP_EXTERNAL_PROJECT`, then rerun the SuperBuild make.

jgrocha, 2018-06-16:
@pierotofy Thanks for the feedback.

I've tried that, but LASzip is not a cmake project. Right now I can download it, configure, but make fails. I'm quite sure I have path related variables not properly set.

```
set(_proj_name laszip)
set(_SB_BINARY_DIR "${SB_BINARY_DIR}/${_proj_name}")

ExternalProject_Add(${_proj_name}
  DEPENDS
  PREFIX            ${_SB_BINARY_DIR}
  TMP_DIR           ${_SB_BINARY_DIR}/tmp
  STAMP_DIR         ${_SB_BINARY_DIR}/stamp
  #--Download step--------------
  DOWNLOAD_DIR      ${SB_DOWNLOAD_DIR}
  URL               https://github.com/LASzip/LASzip/releases/download/v2.2.0/laszip-src-2.2.0.tar.gz
  CONFIGURE_COMMAND <SOURCE_DIR>/configure
  BUILD_COMMAND     make
  #--Install step---------------
  INSTALL_COMMAND   make install
  #--Output logging-------------
  LOG_DOWNLOAD      ON
  LOG_CONFIGURE     ON
  LOG_BUILD         ON
)
```

jgrocha, 2018-06-17:
I was able to do add LASzip and also update PDAL. Now compression is working.
I'm preparing a PR with the changes.

smathermather, 2018-06-18:
👍

closed 2018-06-18 by smathermather

hy9be, 2019-12-27:
Hi all, I am still getting the error ```PDAL: writers.las: Can't write LAZ output.  PDAL not built with LASzip or LAZperf.``` on release 0.9.1. Am I missing anything?

smathermather, 2019-12-30:
Hi @hy9be -- how are you deploying?

hy9be, 2019-12-30:
Hi @smathermather, I am doing a local build. The docker deployment does not have the issue (for sure lol)

smathermather, 2019-12-31:
How are you building PDAL? (Also, we should probably move this over to https://community.opendronemap.org

commit ad4e4264 (2018-06-18): Add support for compressed LAZ files
Add LASzip library to docker
Close #845