OpenDroneMap/ODM #852: control spatial extent of reconstruction

petrasovaa, 2018-06-21:
In case of oblique images ODM tries to reconstruct area way beyond the region of interest. Providing a bounding box or a vector file would, I assume, make reconstruction or at least DSM creation and meshing faster.

smathermather, 2018-06-21:
Ooh, good point. IDK how hard it would be to wrap in with SfM step, but pretty trivial I’d think for meshing.

smathermather, 2020-04-14:
Hmm, still an interesting request. I will keep this open @petrasovaa

kikislater, 2020-04-15:
Grass region computation concept in ODM ! Love the concept ^^

smathermather, 2020-04-15:
:D

pierotofy, 2020-09-14:
I think the photogrammetry way to do this would be to mask the sky regions; it's a more general approach that would work with both geo and non-georeferenced datasets.

smathermather, 2020-09-15:
Masking implemented in 2.0: https://github.com/OpenDroneMap/ODM/pull/1156#issuecomment-692880000

Huge for 360 camera use too.