OpenDroneMap/ODM #854: processing error

dodge-ttu, 2018-06-22:
How did you install WebODM? (Docker, natively, ...)?

I am using open drone map through docker on Ubuntu 18.04

What is the problem?

I'm running ODM through docker on Ubuntu installed on a very powerful desktop computer. I receive the following error:

[DEBUG] running /code/SuperBuild/install/bin/texrecon /code/opensfm/reconstruction.nvm /code/odm_meshing/odm_mesh.ply /code/odm_texturing/odm_textured_model -d gmi -o gauss_clamping -t none
/code/SuperBuild/install/bin/texrecon (built on Jun 4 2018, 13:59:47)
Load and prepare mesh:
PLY Loader: comment PCL generated
Reading PLY: 100058 verts... 199898 faces... done.
Generating texture views:
NVM: Loading file...
NVM: Number of views: 1288
NVM: Number of features: 0
Loading 0%...terminate called after throwing an instance of 'util::Exception'
what(): /code/opensfm/undistorted/batch2: No such file or directory
Aborted (core dumped)
Traceback (most recent call last):
File "/code/run.py", line 47, in 
plasm.execute(niter=1)
File "/code/scripts/mvstex.py", line 139, in process
'{keepUnseenFaces}'.format(**kwargs))
File "/code/opendm/system.py", line 34, in run
raise Exception("Child returned {}".format(retcode))
Exception: Child returned 134

What should be the expected behavior?

The process completes without error as has happened before.

How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

The way to repeat this error would be to use the following parameters when running on the dataset I will try to upload by tomorrow.

sudo docker run -it --rm -v "/home/will/Temp Drone Image Folder/2018-06-21_75_75_20_rainMatrix:/code/images" -v "/home/will/Temp Drone Image Folder/2018-06-21_75_75_20_rainMatrix/odm_orthophoto:/code/odm_orthophoto" opendronemap/opendronemap --orthophoto-resolution 120 --force-ccd 13.2

I will upload the dataset to google drive if possible for testing but it is very large.

dodge-ttu, 2018-06-23:
The problem here, as I finally discovered, is that I my image filenames had characters that caused the process to fail. 

The filenames were batch2 (2).jpg, batch(3).jpg, batch (4).jpg, etc. This is a result of taking over 1000 images with a DJI drone where the naming starts over at 1000 and images are stored in separate folders during flight. I was forced to batch change the filenames so that they could all be in the same folder. Windows always adds a parenthesis and a space which caused the failure.

pierotofy, 2018-06-25:
Duplicate here: https://github.com/OpenDroneMap/WebODM/issues/472

smathermather, 2018-06-26:
Closing as duplicate.

closed 2018-06-26 by smathermather