OpenDroneMap/ODM #856: Memory issues with large dataset

nj07, 2018-06-24:
I installed OpenDroneMap using Docker (Docker version 17.12.0-ce) and running it on Ubuntu.
Operating System: Ubuntu 16.04 LTS
OSType: linux
Architecture: x86_64
CPUs: 12
Total Memory: 23.46GiB

My input size is 375 images. I have tried running OpenDroneMap with and without any additional options. I have also tried it with passing option for 1 processor.

docker run -it --rm -v "$(pwd)/images:/code/images" -v "$(pwd)/odm_prthophoto:/code/odm_orthophoto" -v "$(pwd)/odm_texturing:/code/odm_texturing" opendronemap/opendronemap --opensfm-processes 1 --fast-orthophoto 

 But it always fails:

Load and prepare mesh: 
PLY Loader: comment VTK generated PLY File
Reading PLY: 100075 verts... 200001 faces... done.
Generating texture views: 
NVM: Loading file...
NVM: Number of views: 374
NVM: Number of features: 0
        Loading 100%... done. (Took 79.995s)                                    
Building adjacency graph: 
        Adding edges 100%... done. (Took 0.514s)                                
	299928 total edges.
View selection:
	Building BVH from 200001 faces... done. (Took: 204 ms)
        Calculating face qualities 100%... done. (Took 53.958s)                 
        Postprocessing face infos 100%... done. (Took 0.235s)                   
	Maximum quality of a face within an image: 384729
	Clamping qualities to 11696.9 within normalization.
	Writing data cost file... done.
	Optimizing:
		Time[s]	Energy
		0	203079
		1	201308
		1	198908
		1	197641
		1	197033
		1	196675
		2	196482
		2	196349
		2	196210
		2	196129
		3	196076
		3	196039
		3	196004
		3	195968
	32734 faces have not been seen
	Took: 58.531s
Generating texture patches:
	Running... Killed
Traceback (most recent call last):
  File "/code/run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/mvstex.py", line 139, in process
    '{keepUnseenFaces}'.format(**kwargs))
  File "/code/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 137

It is always running into memory issues. What is the minimum memory required to process  ~400 images. Is there a benchmark?
Please help. Thanks.

pierotofy, 2018-06-25:
The codebase is changing frequently, so we don't have definite benchmarks as of yet (but we have plans to create one in the future). It's not just a function of number of images, but also of resolution. I'd recommend scaling down your images to go past the texturing step (or use WebODM, which does it for you).

kikislater, 2018-06-26:
Does resizing influence focal lens and sensor size ?

kikislater, 2018-06-26:
Thank you Stephen

closed 2018-09-13 by pierotofy