OpenDroneMap/ODM #857: Ubuntu 18.04 Docker 17.06.2 -ce fails to run opendronemap

TheDude005, 2018-06-27:
Docker

Chrome 67 on Linux, Ubuntu 18.04 LTS

When running the example command with images in the images folder I get this output

docker run -it --rm -v "$(pwd)/images:/code/images" -v "$(pwd)/odm_orthophoto:/code/odm_orthophoto" -v "$(pwd)/odm_georeferencing:/code/odm_georeferencing" opendronemap/opendronemap --mesh-size 100000
Unable to find image 'opendronemap/opendronemap:latest' locally
latest: Pulling from opendronemap/opendronemap
281a73dee007: Pull complete 
2aea1b77cff7: Pull complete 
59a714b7d8bf: Pull complete 
0218064da0a9: Pull complete 
ebac621dcea3: Pull complete 
a3ed95caeb02: Pull complete 
b580731643cc: Pull complete 
faa5fbdba239: Pull complete 
1ba72f678208: Pull complete 
Digest: sha256:94fb08f3f19fa1acc098a6909a97e7198b62e892e0ad8e5852ba7158dbbfeadb
Status: Downloaded newer image for opendronemap/opendronemap:latest
Traceback (most recent call last):
  File "/code/run.py", line 8, in <module>
    import ecto
  File "/code/SuperBuild/install/lib/python2.7/dist-packages/ecto/__init__.py", line 30, in <module>
    from ecto.ecto_main import _cell_base, _cell_cpp, __getitem_list__, __getitem_slice__, __getitem_str__, __getitem_tuple__, lookup
ImportError: libecto.so.0.6: cannot open shared object file: No such file or directory

I would assume that the behavior should be that it would install the image into the container and run the program.  All that I can see in Docker is that the image is downloaded but no containers have been created 

I installed Docker through Snap on Ubuntu 18.04 LTS, created a file in my home directory called "images" and populated it with 25 aerial photographs.  I set up my username and password to be part of the Docker group so that I would not have to use sudo and then ran the example command as posted in the Wiki.

pierotofy, 2018-06-27:
The command looks right. Also I just ran a dataset using the docker image, which seems to work on my machine.

I guess I'm not sure?

Make sure your images are in /path/to/dataset/images before running the command.

Then try to mount the container with:

```docker run -ti --rm -v path/to/dataset:/dataset --entrypoint /bin/bash opendronemap/opendronemap```

Then run:

```# ./run.sh --project-path /dataset```

TheDude005, 2018-06-28:
I uninstalled the Snap install of Docker and did the APT repository method instead and the command works.  Not sure why the Snap install would be any different than the apt repo other than maybe a different version of Docker, but that shouldn't affect the image.

pierotofy, 2018-06-28:
I agree, that's odd. If it happens again, please re-open this issue.

closed 2018-06-28 by pierotofy