OpenDroneMap/ODM #858: [Improvement] - Standarize 2D assets resolution parameters, use GSD to estimate max resolution

pierotofy, 2018-06-28:
Problem 1: 

Currently we use less than intuitive ratios for defining the output resolution of orthophotos and DEMs (orthophoto-resolution uses pixels / meter, dem-resolution uses meters / pixel) and both of them fail to account and optimize for ground sampling distance (GSD) estimates.

A classic example is observed when a user decides to fly at high altitude. The total area covered might be really large, but the total achievable resolution might be really low. For an extreme example, if I use 5 satellite images covering a small island, although my highest image resolution (without accounting for overlap) is image width * 5 (say, 5000 * 5 = 15000 pixels wide), but the island covers 50km in one direction, passing the default value of `20` pixels / meter will happily try to create an image 50000 * 20 = 2000000 pixels wide!

Problem 2:

If we have a dataset covering an area in high resolution (say, 0.01 meter / pixel), but the user asks to compute an orthophoto at a lower resolution (say 0.05 meter / pixels, which is the default in `orthophoto-resolution`), mvs_texturing does not take advantage of this information and processes the meshing step using full resolution, only to lose that expensive computation when `odm_orthophoto` is invoked. This is frequently the cause for people not being able to process a dataset without first resizing it.

Proposed solutions:

1) Standarize resolution settings from meters / pixels and pixels / meters to `cm / pixels` (which is used by most other software packages).

2) Compute an average GSD estimate using `GSD = (sensor height (mm) x flight height (m) x 100) / (focal lenght (mm) x image height (pixel))` for all input cameras to provide an upper resolution bound. If the upper bound is met, all 2D outputs should be rendered at this maximum resolution.

3) If the average GSD estimate is higher than the user's requested resolution, resize the images in `opensfm/undistorted` to a value that would match the GSD estimate and scale the focal length value in `reconstruction.nvm` before `mvs_texturing`.

smathermather, 2018-06-28:
Love this. I’ve been thinking about this a long time. It’s nice to see it so clearly articulated in all its parts.

jtinguely, 2018-07-03:
+1. It would be great to standardize the resolution to cm/pixels, it would eliminate some confusion in our software. I am also hoping computing a upper resolution bound will help with our 'out of memory' issues.

pierotofy, 2018-09-16:
This has been implemented and can be closed.

closed 2018-09-16 by pierotofy