OpenDroneMap/ODM #859: Core dump / Error 132 during texturing/meshing with mvs-texturing/texrecon

ichfred, 2018-06-29:
### How did you install OpenDroneMap? (Docker, natively, ...)?

I tried the natively installation on Ubuntu 16.04 as well as a compiled Docker image.

### What's your browser and operating system? 

Serever OS: Ubuntu 16.04

### What is the problem?

During processing a dataset with ODM the process is interrupted on the step of  "Running MVS Texturing Cell"

> [INFO] Running ODM Meshing Cell - Finished
> [INFO] Running MVS Texturing Cell
> [DEBUG] Writing MVS Textured file in: /home/derfred/ODMProjects/TEST001/odm_texturing/odm_textured_model.obj
> [DEBUG] running /home/derfred/OpenDroneMap/SuperBuild/install/bin/texrecon /home/derfred/ODMProjects/TEST001/opensfm/reconstruction.nvm /home/derfred/ODMProjects/TEST001/odm_meshing/odm_mesh.ply /home/derfred/ODMProjects/TEST001/odm_texturing/odm_textured_model -d gmi -o gauss_clamping -t none
> /home/derfred/OpenDroneMap/SuperBuild/install/bin/texrecon (built on Jun 27 2018, 08:31:21)
> Load and prepare mesh:
> PLY Loader: comment PCL generated
> Reading PLY: 100104 verts… 199788 faces… done.
> Generating texture views:
> NVM: Loading file…
> NVM: Number of views: 76
> NVM: Number of features: 0
> **Loading 0%…Illegal instruction (core dumped)**
> Traceback (most recent call last):
> File “OpenDroneMap/run.py”, line 47, in
> plasm.execute(niter=1)
> File “/home/derfred/OpenDroneMap/scripts/mvstex.py”, line 139, in process
> ‘{keepUnseenFaces}’.format(**kwargs))
> File “/home/derfred/OpenDroneMap/opendm/system.py”, line 34, in run
> raise Exception(“Child returned {}”.format(retcode))
> **Exception: Child returned 132**

I'm running the following hardware:

HP proLiant DL385 G6 rack server 
AMD Opteron Processor Model 2435 (6x2.6 GHz) CPU
62GB ram

The returned "Illegal instruction" & "132" may belong to a unsupported hardware issue but it should meets the requirements. The CPU supports the needed features of:  SSE SSE2 SSE3 SSE4a as listed here: https://github.com/OpenDroneMap/node-OpenDroneMap

The issue is also discussed on the ODM Community: http://community.opendronemap.org/t/exit-on-code-132/766 

### What should be the expected behavior? 

texrecon need to be fixed.

### How can we reproduce this? 

Run this setup. The used dataset is the "official" auckerman for testing from the ODM git: https://github.com/OpenDroneMap/odm_data_aukerman/archive/master.zip

fragofer, 2018-08-08:
Hi, here is a workaround to solve this problem:
https://github.com/OpenDroneMap/WebODM/issues/330

pierotofy, 2018-08-09:
This should have been fixed in the latest MVE master. https://github.com/OpenDroneMap/OpenDroneMap/issues/873. We've also changed our fork of mvs-texturing to point to the main MVE repo.

Testing to confirm the problem is solved is welcome.

fragofer, 2018-08-09:
Great news @pierotofy ! We'll test this change later.
However, I think that similar problems could happen again if ODM continues cloning master branches instead of specific tags or releases.

For instance, the project **rayint** (used by mvs-texturing [link](https://github.com/OpenDroneMap/mvs-texturing/blob/master/elibs/CMakeLists.txt)) is cloned using the branch "cuda" which eventually will be merged and deleted. Also, **rayint** is cloning mve from master too (see [link](https://github.com/nmoehrle/rayint/blob/master/elibs/CMakeLists.txt)) causing that a little change in the mve master two libraries may stop working.

I suggest that ODM should fork external libraries in order to minimize the impact of non controlled changes happened in related projects and use tags/releases as much as possible.

Thank you for all! I really find ODM very useful for the community!

pierotofy, 2018-08-09:
`rayint` shouldn't give any problems because it's used by `mvs-texturing` as a header-only library. The MVE clone in that repo is only for building its accompanying test application, which is not invoked in the odm pipeline.

You are right, changes in a related project can cause the ODM build to fail (or worse, cause unpredictable runtime errors).

A good testing framework and infrastructure that builds and runs the entire ODM pipeline on a recurring basis would help uncover these issues more frequently. It's a long term goal.

Making our own forks would also alleviate the problem, at the cost of having to spend time maintaining the forks (which is time consuming).

We would welcome help in any of these areas.

pierotofy, 2018-09-12:
Will close this as the problem seems resolved.

closed 2018-09-12 by pierotofy

ichfred, 2019-04-01:
Once again I have a 132 Error. This time on a different step and on a different hardware. The server CPUs ( 2x AMD Opteron(tm) Processor 6172 12 Core) is not supporting SSSE3. Running the aukerman sample files.

WebODM Settings: "High Resolution" Options: min-num-features: 18000, orthophoto-resolution: 2.0, dem-resolution: 2.0, dsm: true, depthmap-resolution: 1000

`[INFO]    Running ODM Load Dataset Cell
[DEBUG]   Loading dataset from: /www/data/6869d432-05d7-4ca0-92b0-d5bdb7f86404/images
[DEBUG]   /www/data/6869d432-05d7-4ca0-92b0-d5bdb7f86404/images
[...] opensfm extract_metadata
[...] opensfm detect_features
[...] opensfm match_features
[...] opensfm create_traks
[...] opensfm reconstruct
[...] opensfm export_visualsfm 
[...] opensfm undistort
[...] opensfm export_bundler
[...] opensfm export_geocoords
[...] makescene
Processing view view_0019.mve...
Created 77 views with 77 valid cameras.
Imported 77 undistorted images.
[DEBUG]   running /code/SuperBuild/src/elibs/smvs/app/smvsrecon -t24 -a1.0 --max-pixels=1000000 -o1 --debug-lvl=0    /www/data/6869d432-05d7-4ca0-92b0-d5bdb7f86404/smvs
Shading-aware Multi-view Stereo (built on Mar 31 2019, 20:35:53)

Initializing scene with 77 views...
Initialized 77 views (max ID is 76), took 10ms.
Reading Photosynther file (77 cameras, 156068 features)...
Illegal instruction (core dumped)
Traceback (most recent call last):
File "/code/run.py", line 47, in <module>
plasm.execute(niter=1)
File "/code/scripts/smvs.py", line 86, in process
system.run('%s %s %s' % (context.smvs_path, ' '.join(config), tree.smvs))
File "/code/opendm/system.py", line 34, in run
raise Exception("Child returned {}".format(retcode))
Exception: Child returned 132`