OpenDroneMap/ODM #865: Add support for GeoTIFF images

pierotofy, 2018-07-03:
This is a feature request that keeps getting asked for. ODM should support GeoTIFFs along with JPEGs.

pierotofy, 2018-09-03:
Not initially, no.

It has also been asked if raw images should be supported.

jtinguely, 2018-09-24:
What I was hoping this feature would allow me to speed up the generation of a GeoTiff from large datasets. For example, if I have an area to stitch comprised of 2000 jpg's images. I would like to break this large dataset into 4 smaller datasets to produce 4 GeoTiff images approximately 500 jpg's each. Than take the resulting 4 GeoTiff's and stitch together to produce a single GeoTiff image.

dakotabenjamin, 2018-09-24:
@jtinguely that is a slightly different issue which is partially solved already using the experimental "split-merge" process, outlined here: https://github.com/OpenDroneMap/OpenDroneMap/tree/master/scripts/metadataset using a custom split

spifftek70, 2018-11-08:
This is to add support for using *.tif files as input tiles for stitching, correct?  Would this include multispectral images?  RGBN?

pierotofy, 2018-11-08:
Correct.

Long term that would include multispectral images, yes.

JhonyVillacis, 2019-01-13:
Estoy a la espera que se pueda soportar imagenes . tif .tiff para poder usarlo con imagenes multiespectrales en nuestra plataforma de agricultura de precision http://app.agroscan.ec/ , espero no se tarde mucho en soportar este formato.

JhonyVillacis, 2019-01-13:
I'm waiting for images to be supported. tif .tiff to be able to use it with multispectral images in our precision agriculture [platform]( http://app.agroscan.ec/), I hope it does not take long to support this format.

ClaudiaStoecker, 2019-03-15:
Conversion from .tiff to .jpeg always means loss of information (and sometimes even exif data) and thus it would be great if ODM could also support .tiff images as input. We are currently integrating ODM into our data processing workflow of different land data acquisition tools - should we stick to jpeg or can we plan with tiff?

smathermather, 2019-03-15:
If 8-bit works, you can use PNG for now with no loss of fidelity. For 12 or 16-but data however, there is not yet a solution.

smathermather, 2019-03-15:
Also, if tiff is a real need, code or funding contributions are most welcome. 🙏

Manojk07, 2019-12-18:
Any update on adding support for tiff images? 
I am eagerly waiting.

smathermather, 2019-12-18:
Work is underway. Expect an announcement Q1 2020. In the meantime, follow along here: https://community.opendronemap.org/t/tiff-multispectral-support/3232

smathermather, 2019-12-18:
Ah, I see you already are... . :)

pierotofy, 2020-02-03:
Done! Woohoo!

closed 2020-02-03 by pierotofy