OpenDroneMap/ODM #868: TypeError during split-merge

danbjoseph, 2018-07-10:
details posted at: http://community.opendronemap.org/t/typeerror-during-split-merge/808
but couldn't attach the [full log](https://github.com/OpenDroneMap/OpenDroneMap/files/2181746/log-type-error.txt)

_what's the preference for something like this? forum post or GitHub issue?_

```
[94m[INFO]    Generating grammar tables from /usr/lib/python2.7/lib2to3/Grammar.txt[0m
[94m[INFO]    Generating grammar tables from /usr/lib/python2.7/lib2to3/PatternGrammar.txt[0m
[92m[DEBUG]   Found GEOS DLL: <CDLL '/usr/local/lib/python2.7/dist-packages/shapely/.libs/libgeos_c-bd8d3f16.so.1.10.2', handle 2574610 at 7f616ef23dd0>, using it.[0m
[92m[DEBUG]   Trying `CDLL(libc.so.6)`[0m
[92m[DEBUG]   Library path: 'libc.so.6'[0m
[92m[DEBUG]   DLL: <CDLL 'libc.so.6', handle 7f6192ca44e8 at 7f616ef23e10>[0m
Traceback (most recent call last):
  File "/home/ubuntu/OpenDroneMap/scripts/metadataset/merge.py", line 78, in <module>
    if poly.contains(Point(point[[2, 3]])):
  File "/usr/local/lib/python2.7/dist-packages/shapely/geometry/point.py", line 49, in __init__
    self._set_coords(*args)
  File "/usr/local/lib/python2.7/dist-packages/shapely/geometry/point.py", line 130, in _set_coords
    self._geom, self._ndim = geos_point_from_py(args[0])
  File "/usr/local/lib/python2.7/dist-packages/shapely/geometry/point.py", line 209, in geos_point_from_py
    dx = c_double(coords[0])
TypeError: a float is required
```

smathermather, 2018-07-10:
If your hunch is that it is a bug, then by all means post here, though in general terms it is better to start on community and then escalate if needed. This quacks like a bug, so happy to have it here.

We need @dakotabenjamin too look at shapely and the final merge step for split-merge anyway, e.g. : https://github.com/OpenDroneMap/OpenDroneMap/issues/815

I assume all your orthophotos output?

danbjoseph, 2018-07-10:
I checked a few folders for the file via cyberduck (were there) and downloaded and looked at one.
[image]

dakotabenjamin, 2018-07-11:
I'm not sure on this one. Can you tell me what version of `shapely` and `libgeos` you are running?

dakotabenjamin, 2018-07-11:
@smathermather can you also recommend meshing parameters for Dan to improve the ortho there?

danbjoseph, 2018-07-11:
```
$ pip show shapely
---
Metadata-Version: 2.0
Name: Shapely
Version: 1.6.4.post1
Summary: Geometric objects, predicates, and operations
Home-page: https://github.com/Toblerity/Shapely
Author: Sean Gillies
Author-email: sean.gillies@gmail.com
Installer: pip
License: BSD
Location: /usr/local/lib/python2.7/dist-packages
Requires: 
Classifiers:
  Development Status :: 5 - Production/Stable
  Intended Audience :: Developers
  Intended Audience :: Science/Research
  License :: OSI Approved :: BSD License
  Operating System :: OS Independent
  Programming Language :: Python :: 2.6
  Programming Language :: Python :: 2.7
  Programming Language :: Python :: 3
  Topic :: Scientific/Engineering :: GIS
```
 how do i get the `libgeos` version?

smathermather, 2018-07-11:
```ldconfig -v | grep libgeos```

danbjoseph, 2018-07-11:
@dakotabenjamin i can also give you access to the EC2 instance so you can poke around if that would help

danbjoseph, 2018-07-16:
installing on AWS, is there a recommended ami that might have different default software versions? or something i should try to see if this a dependency issue (versions, compatibility, etc)? anything i could try and adjust before running this again?

gyorgytamara, 2018-12-11:
Is there any news on this bug?

pierotofy, 2019-05-23:
Brand new split-merge pipeline is now merged into `master` so this can probably be closed! If there are still errors, please open a new issue. :+1:

closed 2019-05-23 by pierotofy