OpenDroneMap/ODM #869: custom parameter --time -> file location

vgorte, 2018-07-11:
The ODM docs state that the custom parameter --time creates a benchmark file. 
Where can I find this file?
I would really appreciate if someone could help me.

pierotofy, 2018-07-16:
Look for a `benchmark.txt` file.

vgorte, 2018-07-16:
Hey @pierotofy. Thanks for your reply. There just is no benchmark.txt.

I use docker and I ran something like:

```
docker run -it --rm \
    -v $(pwd)/images:/code/images \
    -v $(pwd)/opensfm:/code/opensfm \
    -v $(pwd)/odm_meshing:/code/odm_meshing \
    -v $(pwd)/odm_texturing:/code/odm_texturing \
    -v $(pwd)/odm_orthophoto:/code/odm_orthophoto \
    my_odm_image --time --min-num-features 2500
```
Is there an error in the command?

pierotofy, 2018-07-16:
No, no error, but the file will be created inside the docker container and the file is not mapped to the host filesystem (so it won't show up).

Try with:

`docker run -ti --rm -v $(pwd):/datasets/code my_odm_image --time --min-num-features 2500 --project-path /datasets`

vgorte, 2018-07-16:
Works like a charm. Thank you very much.

closed 2018-07-16 by vgorte