OpenDroneMap/ODM #873: [Improvement] Remove -msse2 -msse3 -msse4 -mpopcnt flags from docker builds

pierotofy, 2018-08-08:
This is currently forcing SSE4 instructions to be included in mvs-texturing (and related mve apps): https://github.com/OpenDroneMap/WebODM/issues/330#issuecomment-411550656

A simple fix would be to filter them out in https://github.com/OpenDroneMap/OpenDroneMap/blob/master/docker/g%2B%2B and https://github.com/OpenDroneMap/OpenDroneMap/blob/master/docker/gcc

People re-building docker images using the standard `Dockerfile` and using older CPUs would still have the problem however.

I'll see if perhaps we can get it fixed directly in MVE.

pierotofy, 2018-08-09:
MVE now ships with -march=native flag so this can be closed. https://github.com/simonfuhrmann/mve/pull/441#event-1779240350

closed 2018-08-09 by pierotofy

nchamo, 2019-05-23:
Hey everyone!

I just wanted to add a comment for people that might still experience the problem when running WebODM with docker in the future. The doc says that you can't run the docker images if you have a processor that doesn't support SSE* instructions. That is actually not true.

With the change made to MVE, the flag "march=native"  adds all optimization flags applicable to the hardware where the code is compiled. So the problem is that when the docker image is created, the SSE* flags that your hardware might not support are actually added, because it was built in a place where they were supported. So instead of building the different repositories natively, you could just re-build the image on your own hardware and that will fix the problem.

I found that easier than installing everything natively.

That's it :smile:

fpolig01, 2020-03-31:
Hey @nchamo I'm running into this issue with my AMD Phenom II X6 1090T CPU. I think your recommendation of rebuilding the image would work for me, however I'm not sure how to do that? Right now I just clone the project and run the startup.sh file. How do I go about rebuilding? Thanks