OpenDroneMap/ODM #876: Cannot compile ODM natively on VMWare ESXi 6.7

pierotofy, 2018-08-20:
Platform specifications: `captive VMWare ESXi 6.7 server. 8 x vcpu and 64GB of memory dedicated`

Error while compiling: ```
[ 55%] Building CXX object libs/tex/CMakeFiles/tex.dir/view_selection.cpp.o
In file included from /code/SuperBuild/src/mvstexturing/elibs/mapmap/mapmap/header/vector_math.h:354:0,
                 from /code/SuperBuild/src/mvstexturing/elibs/mapmap/mapmap/header/costs.h:23,
                 from /code/SuperBuild/src/mvstexturing/elibs/mapmap/mapmap/full.h:15,
                 from /code/SuperBuild/src/mvstexturing/libs/tex/view_selection.cpp:14:
/code/SuperBuild/src/mvstexturing/elibs/mapmap/mapmap/source/vector_math.impl.h: In function ‘void mapmap::v_masked_store(mapmap::_v_t<COSTTYPE, SIMDWIDTH>&, mapmap::_iv_t<COSTTYPE, SIMDWIDTH>&, mapmap::_s_t<COSTTYPE, SIMDWIDTH>*) [with COSTTYPE = float; unsigned int SIMDWIDTH = 4u; mapmap::_v_t<COSTTYPE, SIMDWIDTH> = __vector(4) float; mapmap::_iv_t<COSTTYPE, SIMDWIDTH> = __vector(2) long long int; mapmap::_s_t<COSTTYPE, SIMDWIDTH> = float]’:
/code/SuperBuild/src/mvstexturing/elibs/mapmap/mapmap/source/vector_math.impl.h:1057:33: error: ‘_mm_extract_epi64’ was not declared in this scope
   a1.i = _mm_extract_epi64(aa, 0);
```

pierotofy, 2018-08-20:
Original thread: http://community.opendronemap.org/t/localhost-3000-seems-to-be-offline/932/16

mcevoysean, 2018-08-21:
Processor is a Xeon x5690 with SSE4.2

fredlllll, 2018-11-04:
well its a SIMD instruction "pextrq" (SSE4) which might not be available to the machine in question. even if the host cpu has it, it might not be made available to the guest, or the system didnt recognize it because it only gets the CPUID of the virtual cpu

alke, 2019-10-12:
I change _mm_extract_epi64 to _mm_extract_epi32 where the erros comes

closed 2020-04-14 by smathermather